hku-mars/LiDAR_IMU_Init #1: code release

chengwei0427, 2022-03-14:
Hi，

Great jobs.

Is there any approximate code release date?

zfc-zfc, 2022-03-23:
Hi, thank you for your interest in our work. 
The code would be released once our paper is accepted, the date is around July 25.

Taeyoung96, 2022-05-29:
@zfc-zfc   
Thanks for your great work!  
I have one more question about releasing the code.  
When you release the code, is it released in a merged state into [Fast-LIO](https://github.com/hku-mars/FAST_LIO)?  
Or is the code released as a standalone package or module?  
Thanks,

zfc-zfc, 2022-05-29:
> 
The code of LI_Init will be released as the initialization module of FAST-LIO2, so that FAST-LIO2 can easily run on any LiDAR-IMU combination without dedicated prior calibration or hardware setup. 
LI-Init can also be merged into other LiDAR-inertial odometry algorithms. Since LI-Init is able to provide extrinsic transformation and temporal offset between LiDAR and IMU, other LIO algorithms can easily finish spatial compensation and synchronization using the parameters estimated by LI-Init.

Taeyoung96, 2022-05-30:
@zfc-zfc   
Thanks for your reply.  
Looking forward to some great code!  :smile:

Camilochiang, 2022-07-01:
Hei @zfc-zfc . Does that mean that your lab will incorporate it to fast lío and r3live by default and initial excitation will be recommended or that is that out of the scope of your groups?

thanks!

zfc-zfc, 2022-07-02:
> Hei @zfc-zfc . Does that mean that your lab will incorporate it to fast lío and r3live by default and initial excitation will be recommended or that is that out of the scope of your group?
> 
> thanks!

Hi. The code of LI-Init, which was released yesterday, contains the initialization module and FAST-LIO. If you run the code of LI-Init, it will first do initialization (which tells you the extrinsic transformation and temporal offset) and then it will switch into FAST-LIO. Thus, if you want to run FAST-LIO on your own data but the LiDAR and IMU are not synchronized or calibrated before, you can directly run LI-Init (since it will switch into FAST-LIO after Initialization is finished).
As for R3LIVE, you can write the extrinsic and temporal offset obtained by LI-Init into the config file of R3LIVE.

closed 2022-07-04 by chengwei0427