hku-mars/LiDAR_IMU_Init #3: Question about constant velocity model

qpc001, 2022-07-03:
https://github.com/hku-mars/LiDAR_IMU_Init/blob/2d77c24d29812e033d92ac72211b6165908a3a96/src/IMU_Processing.hpp#L230

`      /** 匀角速度 **/
    M3D Exp_f = Exp(state_inout.bias_g, dt); //用bias_g代表全程不变的角速度
    F_x.block<3, 3>(0, 0) = Exp(state_inout.bias_g, -dt);
    F_x.block<3, 3>(0, 15) = Eye3d * dt;
`

why "F_x.block<3, 3>(0, 15) = Eye3d * dt;" in constant velocity model, but "F_x.block<3,3>(0,15)  = - Eye3d * dt;" in IMU model?

zfc-zfc, 2022-07-04:
> 
In the constant velocity model, bias_g represents the angular velocity. But in IMU model, bias_g represents the bias of gyroscope. Thus the corresponding parts of F_x are different.

zfc-zfc, 2022-07-04:
> https://github.com/hku-mars/LiDAR_IMU_Init/blob/2d77c24d29812e033d92ac72211b6165908a3a96/src/IMU_Processing.hpp#L230
>
> ` /** 匀角速度 **/ M3D Exp_f = Exp(state_inout.bias_g, dt); //用bias_g代表全程不变的角速度 [...]

In the constant velocity model, bias_g represents the angular velocity. But in IMU model, bias_g represents the bias of gyroscope. Thus the corresponding parts of F_x are different.

closed 2022-07-13 by zfc-zfc