hku-mars/LiDAR_IMU_Init #6: 错误：double free or corruption (!prev)

huanghqc, 2022-07-05:
Max Cross-correlation: IMU lag wtr Lidar : 7
Total time delay (IMU wtr Lidar): -9.94719 s
Using LIO: SUBTRACT this value from IMU timestamp
           or ADD this value to LiDAR timestamp.

double free or corruption (!prev)
[laserMapping-2] process has died [pid 25144, exit code -6, cmd /home/hqc/catkin_ts/devel/lib/lidar_imu_init/fastlio_mapping __name:=laserMapping __log:=/home/hqc/.ros/log/af93858c-fc0b-11ec-b4bc-9822ef00c7d8/laserMapping-2.log].
log file: /home/hqc/.ros/log/af93858c-fc0b-11ec-b4bc-9822ef00c7d8/laserMapping-2*.log

zfc-zfc, 2022-07-05:
> @zfc-zfc Max Cross-correlation: IMU lag wtr Lidar : 7 Total time delay (IMU wtr Lidar): -9.94719 s Using LIO: SUBTRACT this value from IMU timestamp or ADD this value to LiDAR timestamp.
> 
> double free or corruption (!prev) [laserMapping-2] process has died [pid 25144, exit code -6, cmd /home/hqc/catkin_ts/devel/lib/lidar_imu_init/fastlio_mapping __name:=laserMapping __log:=/home/hqc/.ros/log/af93858c-fc0b-11ec-b4bc-9822ef00c7d8/laserMapping-2.log]. log file: /home/hqc/.ros/log/af93858c-fc0b-11ec-b4bc-9822ef00c7d8/laserMapping-2*.log

How about trying gdb debug by modifying xxx.launch file like this:
<node pkg="lidar_imu_init" type="fastlio_mapping" name="laserMapping" output="screen" launch-prefix="gdb -ex run --args"/>

zfc-zfc, 2022-07-05:
> @zfc-zfc Max Cross-correlation: IMU lag wtr Lidar : 7 Total time delay (IMU wtr Lidar): -9.94719 s Using LIO: SUBTRACT this value from IMU timestamp or ADD this value to LiDAR timestamp.
>
> double [...]

How about trying gdb debug by modifying xxx.launch file like this (Add launch-prefix="gdb -ex run --args"): 
node pkg="lidar_imu_init" type="fastlio_mapping" name="laserMapping" output="screen" launch-prefix="gdb -ex run --args"

huanghqc, 2022-07-05:
@zfc-zfc 我尝试了很多次，每次的错误信息都不一样，但是好像都和内存溢出、内存释放等有关，您能否通过向日葵远程控制帮我调试一下呢

huanghqc, 2022-07-05:
@zfc-zfc 我使用gdb调试后，报错如下：Thread 1 "fastlio_mapping" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: 没有那个文件或目录.

huanghqc, 2022-07-05:
@zfc-zfc 这个错误和这个链接中所说的很相似https://www.h5w3.com/194576.html，您是否在程序中使用了类似的函数等。

zfc-zfc, 2022-07-05:
能否分享一下出错的数据，我这边测试一下呢？另外，请问你使用的ubuntu版本，gcc版本，雷达型号分别是什么呢？

huanghqc, 2022-07-05:
文件比较大，好像上传不了。ubuntu 18.04    melodic      gcc  7.5.0    VLP16

huanghqc, 2022-07-05:
我们能否通过远程控制软件来解决一下这个问题呢

huanghqc, 2022-07-06:
@zfc-zfc 我在调试你的另一个程序时也出现了Thread 1 "ros_test_node" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: 没有那个文件或目录.

zfc-zfc, 2022-07-07:
> 
请问另一个程序是指哪个程序呢？我们这边目前没有复现你提到的错误，如果你可以提供出错的数据（可以上传google drive 或 onedrive），我们很乐意帮你解决问题。
或者你可以下载我们提供的测试数据，看看能否成功运行？如果能成功运行，可能是数据本身的问题。如果我们的测试数据也报同样的错误，可能就是环境配置上的问题。

huanghqc, 2022-07-07:
@zfc-zfc 另一个程序指的是https://github.com/zfc-zfc/imu_lidar_calibration，也是同样的问题。我的数据在这里，https://1drv.ms/u/s!Ag9T0nETOvFzg1lbkGr89Az2MbGu?e=qRIZUr。你的测试数据中好像没有velodyne的数据包，但是我下载了其他型号的激光雷达的数据，是可以正常运行的，我也怀疑是数据本身的问题，但是并没有找到原因。我的激光雷达是VLP16，IMU是SBG  Ellipse-E。麻烦你帮我看看到底是哪里的问题，已经困扰我两天了。非常感谢你的帮助。

huanghqc, 2022-07-12:
> > @zfc-zfc 另一个程序指的是 [https://github.com/zfc-zfc/imu_lidar_calibration，也是同样的问题。](https://github.com/zfc-zfc/imu_lidar_calibration%EF%BC%8C%E4%B9%9F%E6%98%AF%E5%90%8C%E6%A0%B7%E7%9A%84%E9%97%AE%E9%A2%98%E3%80%82) 我的数据在这里，[https://1drv.ms/u/s!Ag9T0nETOvFzg1lbkGr89Az2MbGu?e=qRIZUr。](https://1drv.ms/u/s!Ag9T0nETOvFzg1lbkGr89Az2MbGu?e=qRIZUr%E3%80%82) 你的测试数据中好像没有velodyne的数据包，但是我下载了其他型号的激光雷达的数据，是可以正常运行的，我也怀疑是数据本身的问题，但是并没有找到原因。我的激光雷达是VLP16，IMU是SBG Ellipse-E。麻烦你帮我看看到底是哪里的问题，已经困扰我两天了。非常感谢你的帮助。
> 
> [image]抱歉，无法下载文件

我身边的人都是可以下载的，要不您试一下这个https://onedrive.live.com/?authkey=%21AFuQavz0DPYxsa4&cid=73F13A1371D2530F&id=73F13A1371D2530F%21473&parId=root&action=locate

huanghqc, 2022-07-19:
> 

你发现我的数据集中的错误了吗

huanghqc, 2022-07-19:
> [image] 抱歉，无法下载文件

你发现我的数据集中的错误了吗

closed 2022-09-09 by zfc-zfc