hku-mars/LiDAR_IMU_Init #7: My setup works with Fast-Lio but not with LIDAR_IMU-Init why?

Rotoslider, 2022-07-09:
I have a Velodyne VLP-16 and a XSens MTI-7dk IMU. (/velodyne_points) (/imu/data). Using Fast_Lio I can see points in Rvis and says IMU connected. I have verified the Points and IMU data is streaming and published. But for some reason with the same settings in LiDAR_IMU_Init nothing is found. Does this package currently only work with Livox? Any suggestions?

zfc-zfc, 2022-07-12:
> 

Hi, theoretically, this package supports both mechanical spinning LiDARs and Solid-state LiDARs. But for mechanical spinning LiDARs, those with more than 32 lines would be better. The low vertical resolution like VLP-16 may lead to bad-quality lidar-odometry.
I have tested Hesai PandarXT (32 lines) and you can find example bags in OneDrive. lidar_type = 1 refers to all Livox lidars, thank you for your correction, so 1 is correct for horizon.
As for "nothing is shown" in rviz, please check the topic name in .yaml file to make sure lidar_imu_init subscribes correct ROS topics.

Rotoslider, 2022-07-13:
Thank you for your response. I started over on a different computer and managed to get the Livox Horizon working and it outputs the results file. 

How do I save the PCD results of the slam. I set the pcd_save_en: value to true but it does not output to a folder.

zfc-zfc, 2022-07-13:
> Thank you for your response. I started over on a different computer and managed to get the Livox Horizon working and it outputs the results file.
> 
> How do I save the PCD results of the slam. I set the pcd_save_en: value to true but it does not output to a folder.

I just fixed PCD saving bug. Please try the new version and remember to enable pcd_save_en in .yaml files.

Rotoslider, 2022-07-13:
Thank you for the quick fix to the save. 
Once I verify the Velodyne is working this weekend I will close this.
Thank you for your quick responses. Much appreciated.

Rotoslider, 2022-07-14:
I have narrowed down the issue why I cannot get LIDAR_IMU_init to work on my portable rig.  It builds fine on my desktop Linux box but I get a bracket error on the pie4 rig. Fast LIo works fine on the pie4. 
The error when building is:
`/home/lipi/ws_livox/src/LiDAR_IMU_Init/src/laserMapping.cpp:1282:1: error: expected ‘}’ at end of input
 1282 | }
      | ^
/home/lipi/ws_livox/src/LiDAR_IMU_Init/src/laserMapping.cpp:772:33: note: to match this ‘{’
  772 | int main(int argc, char **argv) {
      |                                 ^
`
If I add a bracket to the end it build but then nothing shows up. Any suggestion on why this works on my desktop and fist lio works on the pie4 but this package does not?

Rotoslider, 2022-07-17:
I have now tried to get this package to work on 3 different arm computers. Two Pie4's 8gb each and a Jetson AGX. All come up with the same error as above. Ubuntu 18 and 20 and Ros Melodic and Noetic. All same error as above. All thress arm computers work fine with Fast-Lio. With Livox and Velodyne.  Package builds fine on AMD64 machine with Ubuntu 20.04 and Noetic. 
Cmake PCL Eigen are all updated to there latest build on each device. 
If anyone gets working on a arm device I would like to here what you had to do different to get it working.

zfc-zfc, 2022-07-18:
> I have now tried to get this package to work on 3 different arm computers. Two Pie4's 8gb each and a Jetson AGX. All come up with the same error as above. Ubuntu 18 and 20 and Ros Melodic and [...]

Hi, theoretically LI-Init can run on arm computers, since the libraries it depends are nearly the same as FAST-LIO2 (apart from Ceres-Solver). Maybe the code should be done a little modification, but unluckily now we don't have arm computers to test LI-Init. When we have time and device, we will test it and maybe can fix the bug.

WadeFrank, 2022-08-10:
> I have narrowed down the issue why I cannot get LIDAR_IMU_init to work on my portable rig. It builds fine on my desktop Linux box but I get a bracket error on the pie4 rig. Fast LIo works fine on [...]

Meet the same issue on jeston xavier:

/home/slam/LI-Init_ws/src/LiDAR_IMU_Init-main/src/laserMapping.cpp: In function ‘int main(int, char**)’:
/home/slam/LI-Init_ws/src/LiDAR_IMU_Init-main/src/laserMapping.cpp:1281:1: error: expected ‘}’ at end of input
 }
 ^
LiDAR_IMU_Init-main/CMakeFiles/fastlio_mapping.dir/build.make:62: recipe for target 'LiDAR_IMU_Init-main/CMakeFiles/fastlio_mapping.dir/src/laserMapping.cpp.o' failed

Hope someone can give some advice !

WadeFrank, 2022-08-10:
> I have narrowed down the issue why I cannot get LIDAR_IMU_init to work on my portable rig. It builds fine on my desktop Linux box but I get a bracket error on the pie4 rig. Fast LIo works fine on [...]

I think I konw how to fix this bug. 
You just need to change the #endif on line 1166 to line 989. The modified code is as follows: 

` c++
                #ifdef MP_EN
                    omp_set_num_threads(MP_PROC_NUM);
                    #pragma omp parallel for
                 #endif
`

closed 2022-09-09 by zfc-zfc