hku-mars/LiDAR_IMU_Init #8: Can i bypass the IMU Calibration part and use one time generated IMU calibration directly in mapping

rahulsharma11, 2022-07-12:
Hi,

Thanks for the great work. I used Pixhawk and VLP16 lidar and able to run the Lidar_imu_init. Got the Extrinsic file and map generated was decent.

Can i bypass the Calibration part for next run and use the generated calibration file in Fastlio mapping? Apart from Extrinc Rotation and Translation matrix, i need to use other parameters also?

Thanks.

zfc-zfc, 2022-07-13:
> Hi,
>
> Thanks for the great work. I used Pixhawk and VLP16 lidar and able to run the Lidar_imu_init. Got the Extrinsic file and map generated was decent.
>
> Can i bypass the Calibration part for next [...]

Thank you for your tests. The most important parameters for FAST-LIO are the extrinsic rotation and translation matrix, and the time offset. For the same device setup (the relative pose between IMU and LiDAR is fixed), you can just write the extrinsic into FAST-LIO's configuration file.
As for time offset, it depends on the synchronization mechanism of the lidar and IMU. For pixhawk, as far as I know, the timestamp is PC time. If the lidar's timestamp is also PC time, then the time offset might be the same. You can bypass temporal initialization next time.
But for some lidars like Livox avia/horizon, the timestamp origin is the moment when the lidar is powered on. So, if you power it off and power it on again, the timestamp counts from 0. In this condition, temporal initialization is necessary at each time the lidar is powered on.
As for IMU bias and gravity, FAST-LIO can refine them online. You don't need to write these prameters.

huanghqc, 2022-07-14:
> 你好
> 
> 感谢您的出色工作。我使用Pixhawk和VLP16激光雷达，能够运行Lidar_imu_init。得到了外在文件和生成的地图是体面的。
> 
> 我是否可以绕过校准部分进行下一次运行，并在Fastlio映射中使用生成的校准文件？除了Exrinc Rotation和Transpa matrix之外，我还需要使用其他参数吗？
> 
> 谢谢。

您好，我的激光雷达是VLP16，IMU是SBG Ellipse-E，按照要求采集完数据后程序运行出错如下double free or corruption (!prev)
[laserMapping-2] process has died [pid 25144, exit code -6, cmd /home/hqc/catkin_ts/devel/lib/lidar_imu_init/fastlio_mapping __name:=laserMapping __log:=/home/hqc/.ros/log/af93858c-fc0b-11ec-b4bc-9822ef00c7d8/laserMapping-2.log].
log file: /home/hqc/.ros/log/af93858c-fc0b-11ec-b4bc-9822ef00c7d8/laserMapping-2*.log
您能否分享一下您的数据集，以便我找到我的数据集中的错误，谢谢！

closed 2022-09-09 by zfc-zfc