hku-mars/LiDAR_IMU_Init #9: Issue with vlp 16

aditdoshi333, 2022-07-12:
Hello,
Thanks for sharing amazing work. I am able smoothly with livox and pixhawk imu. But whenever I connect vlp 16 it's takes data in 1 2 secs and starts drifting during online refinement phase. 

Please let me know if I had to do some code changes to make it run. 

Thank you

zfc-zfc, 2022-07-13:
I'm sorry I didn't tested lidar_imu_init with VLP-16, BUT this guy has successfully run lidar_imu_init on VLP-16 and Pixhawk #8 
Maybe he can give us some useful suggestions.

rahulsharma11, 2022-07-13:
Hi,
Thanks for referring me here. I would like to add-
1) The calibration happened is so unexpected that it only happened once. After that i am not able to calibrate it.
2) I tried different orientation of Pixhawk also but no result.
3) As soon i start rotating in along any axis slightly, all the excitations in x,y and z got quickly filled and it starts online refinement and the calibration gets screwed.
4) Same pixhawk works well with Livox setup.

That way i am not confident about VLP16 with external IMU.

Is there any way we can use Odometry (like from realsense T265) directly with the Velodyne in Fast-Lio? or any other algorithm you can suggest.

Thanks.

rahulsharma11, 2022-07-14:
Hi, 
Thanks for the suggestions.

I tried all your above mentioned points. But no Luck.
Also i tried VoxleMap, Over there either i use IMU or not, movement of Lidar does not produce Odoemtry. Also point cloud is not that good as FastLio(Even using Identity Matrix as Extrinsic Calibration). 

My query is still there -
To do SLAM with VLP16 , i need external IMU. Also for longer run i need good Lidar-IMU Calibration and that is not happening correctly.

Taeyoung96, 2022-08-18:
@rahulsharma11   
Hi, did you solve the problem?

shaoqinshi, 2022-09-08:
I have the same question, how it works for 16 scans lidar

af-doom, 2022-10-15:
Hello, have you solved the calibration problem of velodyne16

af-doom, 2022-10-16:
> @rahulsharma11 Hi, did you solve the problem?

Hello, have you solved the calibration problem of velodyne16

af-doom, 2022-10-16:
> Hello, have you solved the calibration problem of velodyne16

Hello, have you solved the calibration problem of velodyne16

zfc-zfc, 2022-10-18:
Please see an example of successfully calibrating VLP16 and external IMU in #35