hku-mars/LiDAR_IMU_Init #17: Question about implementation of `Forward_propagation_without_imu` function

Taeyoung96, 2022-08-03:
@zfc-zfc   @Joanna-HE  
Thanks for your wonderful works!  

I want to apply the initialization method presented in [LiDAR_IMU_Init](https://github.com/hku-mars/LiDAR_IMU_Init) directly to [FAST-LIO](https://github.com/hku-mars/FAST_LIO) algorithm.  

But I was confused because the implementation is a little bit different. :cry:   

To express the state vector, LiDAR_IMU_Init used `struct StatesGroup` but FAST-LIO used `IKFoM_toolkit`.  

I'm not familiar with [IKFoM](https://github.com/hku-mars/IKFoM), so I don't have an idea to implement [ImuProcess::Forward_propagation_without_imu()](https://github.com/hku-mars/LiDAR_IMU_Init/blob/c98c18ac108e0408c039aaf8cc9e19cec8e0ffcc/src/IMU_Processing.hpp#L204) function to FAST-LIO.  

Would you mind giving me some tips?  

In LiDAR_IMU_Init, `StateGroup &state_inout` and
in FAST-LIO `esekform::esekf<state_ikfom, 12, input_ikfom> &kf_state` represent the same state vector?  

What does the variable in `kf_state` correspond to the `.cov` of `StateGroup`?
 
I'd appreciate your help.  :bow:  

Thanks,

zfc-zfc, 2022-08-12:
Hi. LiDAR_IMU_Init contains not only a full initialization module, but also the whole FAST-LIO2 (directly implemented by Eigen, not ikfom, but they are the same actually). In LiDAR_IMU_Init, it will first do initialization. After initialization finished, it will switch into FAST-LIO2 to refine extrinsic transformation of LiDAR and IMU. So, you don't need to apply initialization module to FAST-LIO algorithm because that has already been finished by LiDAR_IMU_Init.

Taeyoung96, 2022-08-12:
@zfc-zfc   
I understand it! Thanks for your reply :)

closed 2022-08-12 by Taeyoung96