hku-mars/LiDAR_IMU_Init #19: LiDAR-only odometry is failed with M2DGR dataset

Taeyoung96, 2022-08-19:
@zfc-zfc   
Thanks for your great work.  
When I use LiDAR-IMU init with my custom data, LiDAR-only odometry is failed.  

I use the [M2DGR dataset](https://github.com/SJTU-ViSYS/M2DGR) (Velodnye VLP-32 and handsfree IMU)

My custom m2dgr yaml file is like below.  
```
common:
    lid_topic:  "/velodyne_points"
    imu_topic:  "/handsfree/imu"

preprocess:
    lidar_type: 2                # Velodyne LiDAR
    scan_line: 32
    blind: 2
    feature_extract_en: true

initialization:
    cut_frame: true # false: do not cut input pointcloud
    cut_frame_num: 3 # must be positive integer
    orig_odom_freq: 10
    mean_acc_norm: 9.81 # 1: for livox built-in IMU
    online_refine_time: 15.0
    data_accum_length: 200
    Rot_LI_cov: [ 0.00005, 0.00005, 0.00005 ]
    Trans_LI_cov: [ 0.0001, 0.0001, 0.0001 ]

mapping:
    filter_size_surf: 0.5
    filter_size_map: 0.5
    gyr_cov: 10
    acc_cov: 2
    b_acc_cov: 0.0001
    b_gyr_cov: 0.0001
    det_range: 200.0

publish:
    path_en:  true
    scan_publish_en:  false       # false: close all the point cloud output
    dense_publish_en: false       # false: low down the points number in a global-frame point clouds scan.
    scan_bodyframe_pub_en: true  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: false
    interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.
```

And the result looks like this, I used the `door_02` sequence. (outdoor scene) 
[image]

LiDAR-only odometry is failed so online refinement is inaccurate.  

Can you tell me what caused this?  
Do I need to modify the config file?
I saw this issue #11 , is it impossible to calculate lidar-only odometry measured by Ground robot?

I'd appreciate it if you give me an answer.  
Thanks,

zfc-zfc, 2022-08-22:
> lidar-only odometry measured by Ground robot

Sorry for the late reply. I will download this data and try to find out why lidar odometry failed.
Theoretically, this lidar odometry (FAST-LO) should support ground vehicles. But LiDAR-IMU initialization cannot be well achieved on ground vehicles since there lacks excitation of some degrees.

zfc-zfc, 2022-08-22:
> feature_extract_en

I suggest feature_extract_en is set to false (even though it might not be the cause of odometry failure).

Taeyoung96, 2022-08-23:
@zfc-zfc   
Thanks for your reply. I'll try again based on your advice.    
If you succeed with the M2DGR dataset, please let me know. :smile:   
Thanks,

zfc-zfc, 2022-08-26:
> @zfc-zfc Thanks for your reply. I'll try again based on your advice. If you succeed with the M2DGR dataset, please let me know. smile Thanks,

Hi, the problem is clear now. The offset time of each point given by this bag is abnormal. By printing the time field of the input point, I saw some negative values. So I have to compute the offset time by hands. The updated version can run on the M2DGR dataset, at least for the first 50 seconds (The bag I downloaded is incomplete).
[image]

Warm Reminder: If you just want to run LiDAR odometry, not want to use lidar imu initialization, you can set data_accum_length as 9999999, and decrease gyr_cov and acc_cov to 0.1~1.0.

Taeyoung96, 2022-08-27:
@zfc-zfc   
Thanks for your update!  
I tested the updated version on the Docker container that I PR in this repository. :smile:  
Thanks,  
[image]

closed 2022-08-27 by Taeyoung96