hku-mars/LiDAR_IMU_Init #21: Livox config explanation

ManChrys, 2022-08-22:
Hello guys,

I would like to ask if there is an explanation about all parameters, which are included in avia.yaml. I have a livox mid 40 and I would like to convert the avia.yaml file to a mid 40 compatible file .

Thanks in advance.

zfc-zfc, 2022-08-22:
> Hello guys,
> 
> I would like to ask if there is an explanation about all parameters, which are included in avia.yaml. I have a livox mid 40 and I would like to convert the avia.yaml file to a mid 40 compatible file .
> 
> Thanks in advance.

Please read README.md carefully. Thanks.

ManChrys, 2022-08-22:
Thanks for the fast response . 

i saw the "important parameters" paragraph but i didnt see anything about

_**1st question : how can i find these params for mid 40 ?**_

    feature_extract_en: false
    scan_line: 6
    blind: 2

and

**_2nd question : these are my imu parameters grom manual ?_**

    gyr_cov: 50
    acc_cov: 2
    b_acc_cov: 0.0001
    b_gyr_cov: 0.0001
    det_range: 450.0

also

**_3rd question : rotation and translation are between lidar and imu ?_** 

    Rot_LI_cov: [ 0.00005, 0.00005, 0.00005 ]
    Trans_LI_cov: [ 0.00001, 0.00001, 0.00001 ]

zfc-zfc, 2022-08-22:
> Thanks for the fast response .
> 
> i saw the "important parameters" paragraph but i didnt see anything about
> 
> _**1st question : how can i find these params for mid 40 ?**_
> 
> ```
> feature_extract_en: false
> scan_line: 6
> blind: 2
> ```
> 
> and
> 
> **_2nd question : these are my imu parameters from manual ? I have xsens mti-g-710 gnss_**
> 
> ```
> gyr_cov: 50
> acc_cov: 2
> b_acc_cov: 0.0001
> b_gyr_cov: 0.0001
> det_range: 450.0
> ```
> 
> also
> 
> **_3rd question : rotation and translation are between lidar and imu ?_**
> 
> ```
> Rot_LI_cov: [ 0.00005, 0.00005, 0.00005 ]
> Trans_LI_cov: [ 0.00001, 0.00001, 0.00001 ]
> ```
For the first question:
[image]
I suggest the parameters in the 2nd and 3rd question remain unchanged. 
gyr_cov, acc_cov, b_acc_cov, b_gyr_cov, Rot_LI_cov, Trans_LI_cov are propagation noise of angular velocity, linear acceleration, gyroscope bias and accelerometer bias, extrinsic rotation, extrinsic translation.

ManChrys, 2022-08-22:
Thanks a lot my friend!!! Have a nice day!!

closed 2022-08-22 by ManChrys

hr2894235132, 2022-10-10:
> Thanks for the fast response .
>
> i saw the "important parameters" paragraph but i didnt see anything about
>
> _**1st question : how can i find these params for mid 40 ?**_
>
> ```
> feature_extract_en: [...]

hey，what about the first question? Can you teach me?