hku-mars/LiDAR_IMU_Init #22: "No IMU, use constant velocity model."

ManChrys, 2022-08-25:
Hello,

i've tried the package but when i run it the screen is black
[image]

and on cmd windows says "No IMU, use constant velocity model."
[image]

I tested every unit individual (livox mid 40 and xsens mti g 710 gnss/ins) throught their softwares and ros drivers and they are working properly.

in my yaml i just changed the imu topic as you can see below.

[image]

Thanks in advance.

zfc-zfc, 2022-08-25:
> as

The screen is black because li_init does not subscribe lidar or imu topic.
Please use rostopic echo xxx to make sure the correct topic is published. If there is nothing when you rostopic echo /livox/lidar, it seems the livox_ros_driver does not work properly. 
By the way, what does this mean "Further more if i try to run livox_ros_driver it closed automatically when i run the livox_avia.launch file, **because its the same node**"?

ManChrys, 2022-08-25:
Hello @zfc-zfc and thanks for the fast answer again!!

When i use livox mid 40 and run the `roslaunch livox_ros_driver livox_lidar.launch` it is working properly. I use `rostopic echo /livox/lidar ` and i am seeing the measurements.

When i use xsens mti-g-710 gnss/ins (imu) and run the `roslaunch xsens_mti_driver example.launch` command imu is working properly i have all meauserments (angular, acc etc), i saw them using the `rostopic echo /imu/data` 

[image]

When i try to run lil_init project i have the problems that i mention above :/

Am i missing something ? :/

Thanks in advance.

zfc-zfc, 2022-08-25:
> above

So you said, when you run "roslaunch livox_ros_driver livox_lidar_msg.launch" and "roslaunch xsens_mti_driver example.launch", both /livox/lidar and /imu/data topics are published, you can see data after "rostopic echo xxx". 
But after you run LI-Init, /livox/lidar disappears but /imu/data still works normally?

zfc-zfc, 2022-08-25:
> Should i run livox_ros_driver to a terminal, xsens_mti_ros_driver to another terminal and lil_init to a third terminal

Yes you should run livox_ros_driver to a terminal, xsens_mti_ros_driver to another terminal and li_init to a third terminal.

ManChrys, 2022-08-25:
> So you said, when you run "roslaunch livox_ros_driver livox_lidar_msg.launch" and "roslaunch xsens_mti_driver example.launch", both /livox/lidar and /imu/data topics are published, you can see data [...]

exactly!!

> Yes you should run livox_ros_driver to a terminal, xsens_mti_ros_driver to another terminal and li_init to a third terminal.
I tried it but when im launching lil_init, the terminal in which i am launching livox_ros_driver shut down and say something like that "livox ros driver stops because a same node is running" with red letters (i am not on my pc right now to send you the exactly message)

zfc-zfc, 2022-08-25:
> unavailable

It's strange. Maybe you should check what is the name of the same node. As far as I know, LI-Init cannot have any conflicts with livox_ros_driver because we have tested it hundreds of times.

ManChrys, 2022-08-25:
Yeah i know its really strange. Ill try it again tomorrow and if the promblem remain ill post here the screenshots.

Thanks for the fast response i really appreciate it.

pingguofu, 2022-08-26:
Hello,
Has this problem been solved? I used the IMU of xsens and encountered the same problem.

zfc-zfc, 2022-08-26:
> remains

How about try this: Only launch livox_ros_driver and IMU driver, record a bag and make sure this bag contain correct data (rosbag info xxx.bag). Then shut down the drivers, only run lidar_imu_init, then rosbag play xxx.bag.

ManChrys, 2022-08-26:
Using your project you can calibrate LiDAR and imu in the beginning right ? If they are not synchronised how can I calibrate them using a bag file ?

ManChrys, 2022-08-26:
> Hello,
> 
> Has this problem been solved? I used the IMU of xsens and encountered the same problem.
> 
> 
> 
> 

I'll be able to try it again in 2-3 hours from now . If I solve it I'll inform you .

zfc-zfc, 2022-08-26:
> calibrate

Of course you can calibrate them using a bag file, otherwise why would I upload example bags?

ManChrys, 2022-08-26:
> > calibrate
> 
> 
> 
> Of course you can calibrate them using a bag file, otherwise why would I upload example bags?

Yeah that's right . I'll try it in about 2-3 hours and I'll inform you 😃

ManChrys, 2022-08-26:
Well i have to say the following things.

When im running  `roslaunch livox_ros_driver livox_lidar.launch` !!!alone!!! i have the following
[image]

When im running `roslaunch livox_ros_driver livox_lidar_msg.launch` !!!alone!!! i have the following
[image]

When im running `roslaunch livox_ros_driver livox_lidar.launch` with `roslaunch lidar_imu_init livox_avia.launch` i have the following
[image]

When im running `roslaunch xsens_mti_driver xsens_mti_node.launch` with `roslaunch lidar_imu_init livox_avia.launch` i have the following
[image]

Also i found that my imu even if is get recognized for some reason when im using `rostopic echo /imu/data` ii doesnt show me something. Using MT manager (imu's software) i have data.
[image]

So im really confused :'D

zfc-zfc, 2022-08-26:
> 

My bro, you should learn some basic knowledge about ROS (ros message and ros topic). Clearly, LI-Init wants to subscribe /livox/lidar with customized message format, while the livox_ros_driver you launched just publish /livox/lidar with pointcloud2 format. 
[image]

If you read the README of livox_ros_driver, you should notice there are two launch files given: 
[image]

And in my previous answer, I have already said you should run **livox_lidar_msg.launch**.
[image]

Maybe next time, you could firstly try to analyze the reason you failed by yourself. If the reason is truly caused by others not yourself, then ask others.

ManChrys, 2022-08-26:
I fixed the problem with red letters

`[ERROR] [1661495465.896352538]: Client [/laserMapping] wants topic /livox/lidar to have datatype/md5sum [livox_ros_driver/CustomMsg/e4d6829bdfe657cb6c21a746c86b21a6], but our version has [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181]. Dropping connection.`

just change in livox_lidar.launch file the <arg name="xfer_format" default="0"/> to <arg name="xfer_format" default="1"/>

ManChrys, 2022-08-26:
@zfc-zfc thank you for your time and help.

@pingguofu now it worked fine. imu model xsens mti - g - 710 gnss/ins

Ok the whole problem caused from the IMU driver and the custom pcb i have made to convert rs232 to ttl . I reinstall the imu drivers and i used a new pcb and everything is running smooth. 

[image]

closed 2022-08-26 by ManChrys