hku-mars/LiDAR_IMU_Init #23: datatype/md5sum error

Congxilong, 2022-08-29:
[ERROR] [1661737358.718783357]: Client [/laserMapping] wants topic /livox/lidar to have datatype/md5sum [livox_ros_driver/CustomMsg/e4d6829bdfe657cb6c21a746c86b21a6], but our version has [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181]. Dropping connection.

When I finish compiling and running _Livox-mid360. launch_ and play my bag,the above error occurred.How to solve this problem?thanks!

zfc-zfc, 2022-08-29:
> solve

It seems the /livox/lidar topic in your bag is pointcloud2 format. Please run **roslaunch livox_ros_driver livox_lidar_msg.launch** to get customized format. 
[image]

Congxilong, 2022-08-29:
> > 解决
> 
> 似乎你包里的/livox/lidar主题是pointcloud2格式。请运行 **roslaunch livox_ros_driver livox_lidar_msg.launch** 以获取自定义格式。 [image]

Thanks,I solve my problem by this way,but the result is not satisfactory . I have tried both livox-mid360 and velodyne-VLP16 ,but after a while the map doesn't match . Do I need to modify cut_frame_num or  data_accum_length in YAML file ? Or what do I need to pay attention to when I record packets? Thanks for your time !

zfc-zfc, 2022-08-29:
> 

Livox mid360 should be easy to succeed. It is recommended to give sufficient excitation following the .GIF in README.
[image]

closed 2022-09-09 by zfc-zfc