hku-mars/LiDAR_IMU_Init #24: If I cannot move the vehicle at a large Angle, can I calibrate it?

Congxilong, 2022-08-30:
As the title ，the vehicle 's lidar and IMU positions are fixed , and it can only rotate about the Z axis. As for X or Y axis , it can only rotate about ten degrees. Is the algorithm feasible in this case ? Thanks !

Congxilong, 2022-08-30:
Also, how much does it matter that the IMU is only 100Hz?

Taeyoung96, 2022-08-30:
@Congxilong   
You could fix the function `LI_Init::data_sufficiency_assess()`  
[This line](https://github.com/hku-mars/LiDAR_IMU_Init/blob/243d38febdb77311d250ddb085eabd3bc76f0b02/include/LI_init/LI_init.cpp#L553) decided how to move device to obtain sufficient excitation.  
For example, change `0.99` to `0.1` to obtain short excitation, however, it affects to calibration result.  
In my experience, you have to accept inaccurate calibration results.

zfc-zfc, 2022-08-30:
> Also, how much does it matter that the IMU is only 100Hz?

100Hz is OK. But as Taeyoung said, sufficient excitation around all degrees is necessary for accurate calibration result.

Congxilong, 2022-08-31:
> > Also, how much does it matter that the IMU is only 100Hz?
> 
> 100Hz is OK. But as Taeyoung said, sufficient excitation around all degrees is necessary for accurate calibration result.

I changed the code according to what Taeyoung said, but the result is not very ideal, but thank you anyway !

Congxilong, 2022-08-31:
> > Also, how much does it matter that the IMU is only 100Hz?
> 
> 100Hz is OK. But as Taeyoung said, sufficient excitation around all degrees is necessary for accurate calibration result.

Is it suitable for a six-axis IMU? I just realized the problem : (

closed 2022-10-11 by zfc-zfc