hku-mars/LiDAR_IMU_Init #25: When my program was running, it was stuck in this log

wzszds, 2022-09-05:
[Initialization] Rotation around Lidar X Axis:   0% [                              ]
[Initialization] Rotation around Lidar Y Axis:   0% [                              ]
[Initialization] Rotation around Lidar Z Axis:   0% [                              ]

When my program was running, it was stuck in this log.  How should I troubleshoot the problem？

zfc-zfc, 2022-09-06:
> troubleshoot
what is the type of the LiDAR you used? Did you give sufficient excitation following README?

zfc-zfc, 2022-09-08:
> 正常情况下六轴IMU即可。请问rviz中雷达里程计是否正常运行，点云地图是否清晰一致呢？可以尝试增加设备运动的幅度，使得激励更大一些。请问命令行是否弹出过“[Initialization] Movement detected, data accumulation starts.”这句话呢？

wzszds, 2022-09-09:
我看到rviz里面的里程计是正常的，可以看到车的行驶轨迹，点云地图和附近的点云场景我也能看的清楚。比如旁边有行驶过去的车啊，停着的车啊。我看到了这个log:
Acceleration norm  : 9.80027

[Initialization] Movement detected, data accumulation starts.
 我们的机械式激光雷达是装在车顶的，imu安装在车后备箱里面，我们采集的数据场景是一个泊车的场景，车先往前开几米，然后倒车入库到旁边的一个车库里面。这个github的Issues好像不能添加图片，不然我截图，附在评论里面给你看看rviz里面的点云地图和里程计轨迹，以及我们的运动场景是什么样的。或者您觉得我们应该采集一个什么样的场景的数据，然后可以适合用来做这个雷达跟imu的标定。您可以说车要怎么开，然后怎么开，等等，然后我们就去录这样的rosbag包。

zfc-zfc, 2022-09-09:
> 我看到rviz里面的里程计是正常的，可以看到车的行驶轨迹，点云地图和附近的点云场景我也能看的清楚。比如旁边有行驶过去的车啊，停着的车啊。我看到了这个log: Acceleration norm : 9.80027
>
> [Initialization] Movement detected, data accumulation starts. [...]

由于这个算法需要绕各轴的激励，尤其是旋转，目前这个算法不适用于车这种只能沿平面运动的设备（在issue11中提到过）。
所以应该是因为没有绕各轴的旋转所以导致激励是０

closed 2022-09-15 by zfc-zfc