hku-mars/LiDAR_IMU_Init #35: Hello, I have the following problems when calibrating xsens and velodyne16

af-doom, 2022-10-15:
[image]

af-doom, 2022-10-17:
Hello, I still haven't solved this problem. There is no such error when the data package runs Fast-LIO2

zfc-zfc, 2022-10-17:
> Hello, I still haven't solved this problem. There is no such error when the data package runs Fast-LIO2

Hi, are you convenient to share the bag that the error occurred?

af-doom, 2022-10-17:
> > Hello, I still haven't solved this problem. There is no such error when the data package runs Fast-LIO2
> 
> Hi, are you convenient to share the bag that the error occurred?

Hello, thank you very much for your reply.  The bag has been sent to your mailbox.  Thank you very much.

zfc-zfc, 2022-10-18:
> > > Hello, I still haven't solved this problem. There is no such error when the data package runs Fast-LIO2
> > 
> > 
> > Hi, are you convenient to share the bag that the error occurred?
> 
> Hello, thank you very much for your reply. The bag has been sent to your mailbox. Thank you very much.

Hi, I didn't receive the email. How about sending it to zhufc@connect.hku.hk again, or upload your bag to onedrive and share a link?

af-doom, 2022-10-18:
> > > > Hello, I still haven't solved this problem. There is no such error when the data package runs Fast-LIO2
> > > 
> > > 
> > > Hi, are you convenient to share the bag that the error occurred?
> > 
> > 
> > Hello, thank you very much for your reply. The bag has been sent to your mailbox. Thank you very much.
> 
> Hi, I didn't receive the email. How about sending it to [zhufc@connect.hku.hk](mailto:zhufc@connect.hku.hk) again, or upload your bag to onedrive and share a link?

Thank you very much. I have installed your email address and resend it

af-doom, 2022-10-18:
Thank you very much. I wish I were next to the computer now. I'll see it soon. Could you tell me what the problem is

      

  
  
  
Hi, I didn't receive the email. How about sending it to ***@***.*** again, or upload your bag to onedrive and share a link?
  
 The bug is fixed. Please try the newest version. Thanks.

zfc-zfc, 2022-10-18:
> Thank you very much. I wish I were next to the computer now. I'll see it soon. Could you tell me what the problem is
> […](#)

Some points of Velodyne16 are invalid, the coordinate value is nan. These points should be discarded but I ignored this thing before. The new version is tested OK now.

af-doom, 2022-10-18:
OK, thank you very much for your answer

---Original---
From: "Fangcheng ***@***.***&gt;
Date: Tue, Oct 18, 2022 20:00 PM
To: ***@***.***&gt;;
Cc: ***@***.******@***.***&gt;;
Subject: Re: [hku-mars/LiDAR_IMU_Init] Hello, I have the following problemswhen calibrating xsens and velodyne16 (Issue #35)

closed 2022-11-09 by af-doom

HamzaCetin0, 2024-07-25:
Hi @af-doom. How did you solve the problem. I have the same IMu and Lidar. I encountered the same issue. After the initialization starts the laser scan drifts and then it gives the same error like you got. Can you help me please?