hku-mars/LiDAR_IMU_Init #37: 标定建图后地面有很大的倾斜角度

18954534983, 2022-10-27:
您好，感谢您出色的工作。有个问题想要咨询下您，我用LI-Init标定后，用FAST-LIO建图，整个地面是倾斜的，轨迹在z轴方向持续上扬，并且这个问题在其他slam算法中也存在，请问这是lidar与imu没有标定好的原因吗，还是有其他的原因，这个问题困扰我很久了，我尝试过各种方法，但一直没有解决，非常非常期待您的回复。

liang0724s, 2022-10-28:
> 您好，感谢您出色的工作。有个问题想要咨询下您，我用LI-Init标定后，用FAST-LIO建图，整个地面是倾斜的，轨迹在z轴方向持续上扬，并且这个问题在其他slam算法中也存在，请问这是lidar与imu没有标定好的原因吗，还是有其他的原因，这个问题困扰我很久了，我尝试过各种方法，但一直没有解决，非常非常期待您的回复。

可以从下面这个知乎帖的答案中寻找一些思路
[纯Lidar SLAM如何优化解决Z轴方向的偏移？](https://www.zhihu.com/question/533056547)

18954534983, 2022-11-03:
> > 您好，感谢您出色的工作。有个问题想要咨询下您，我用LI-Init标定后，用FAST-LIO建图，整个地面是倾斜的，轨迹在z轴方向持续上扬，并且这个问题在其他slam算法中也存在，请问这是lidar与imu没有标定好的原因吗，还是有其他的原因，这个问题困扰我很久了，我尝试过各种方法，但一直没有解决，非常非常期待您的回复。
> 
> 可以从下面这个知乎帖的答案中寻找一些思路 [纯Lidar SLAM如何优化解决Z轴方向的偏移？](https://www.zhihu.com/question/533056547)

感谢，您提供的这份资料对我很有帮助，我会尝试使用添加地面约束的方法解决z轴方向漂移的问题，但是资料上也说明了使用地面约束对有坡度的场景不太友好，但我仍然会先尝试一下这个方法。

closed 2022-11-26 by zfc-zfc