hku-mars/LiDAR_IMU_Init #38: Avia lidar ,Data accumulation finished, but ”Max Cross-correlation: IMU lag wtr Lidar : 47 corrupted size vs. prev_size [laserMapping-1] process has died“

skyengine83, 2022-10-31:
I use avia lidar , but dose not get initialization result .
[263d2f690ff238d39ef8f3b4ecbbf8c](https://user-images.githubusercontent.com/13100641/198927551-bfe14bf9-95df-4586-b4e8-4ba6588829e5.png)

skyengine83, 2022-11-03:
Lidar IMU initialization begins but doesn't finished.

xaioqi, 2023-05-04:
Hello, I also encountered the same problem, how did you solve it? ? Thanks

closed 2023-07-12 by zfc-zfc

nes9966, 2023-12-08:
Did you ever solve this issue?

seungjae99, 2024-08-16:
@zfc-zfc 
Hi, thanks for sharing your great work, but I'm also facing the same issue as mentioned above. Could you please provide a solution?

[Initialization] Rotation around Lidar X Axis: 100% [||||||||||||||||||||||||||||||]
[Initialization] Rotation around Lidar Y Axis: 100% [||||||||||||||||||||||||||||||]
[Initialization] Rotation around Lidar Z Axis: 100% [||||||||||||||||||||||||||||||]
[Initialization] Data accumulation finished, Lidar IMU initialization begins.

============================================================ 

Max Cross-correlation: IMU lag wtr Lidar : -778
[laserMapping-1] process has died [pid 20464, exit code -11, cmd /home/catkin_ws/devel/lib/lidar_imu_init/li_init __name:=laserMapping __log:=/root/.ros/log/ce6da3c6-5b12-11ef-8d79-dcfe07109cf5/laserMapping-1.log].
log file: /root/.ros/log/ce6da3c6-5b12-11ef-8d79-dcfe07109cf5/laserMapping-1*.log
^C[rviz-2] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

if you need my bag file i can share it with you

thanks