hku-mars/LiDAR_IMU_Init #39: 标定效果良好，应用到fast-lio是漂移（使用标定时的数据包也漂移）

af-doom, 2022-11-04:
您好，我的IMU是MTI-300，雷达是velodyne16，下面是标定过程和应用到fast-lio2的截图
[image]
[image]
/media/wyw/8660E1BB60E1B25B/2022-11-04-08-43-24.bag

zfc-zfc, 2022-11-04:
> 您好，我的IMU是MTI-300，雷达是velodyne16，下面是标定过程和应用到fast-lio2的截图，第一张图是应用到fast-lio2,第二张图是标定过程（效果很好）,我录制的bag包的链接如下 链接: https://pan.baidu.com/s/1jT_9axsTvxiLaaURQb1AEA 提取码: 1234 复制这段内容后打开百度网盘手机App，操作更方便哦 [image] [image] /media/wyw/8660E1BB60E1B25B/2022-11-04-08-43-24.bag

我这能跑啊，也不漂。**标出来的时间差是-0.308，你填的是-0.019**，是不是这填错了？
b_acc_cov和b_gyr_cov这些最好用默认的0.0001（不过影响应该不大）。
[image]

zfc-zfc, 2022-11-04:
还有一个小建议是，给的数据的场景太狭窄了，刚开始几帧很大的FoV都被桌子挡住，有效点太少，不利于建立地图。可以换个开阔点的地方，或者把雷达放高一点，盲区内尽量不要有遮挡。

af-doom, 2022-11-04:
> 还有一个小建议是，给的数据的场景太狭窄了，刚开始几帧很大的FoV都被桌子挡住，有效点太少，不利于建立地图。可以换个开阔点的地方，或者把雷达放高一点，盲区内尽量不要有遮挡。
[image]

非常感谢您的回复，我修改了相关参数，但是效果好像和您的有较大差异

zfc-zfc, 2022-11-04:
> > 还有一个小建议是，给的数据的场景太狭窄了，刚开始几帧很大的FoV都被桌子挡住，有效点太少，不利于建立地图。可以换个开阔点的地方，或者把雷达放高一点，盲区内尽量不要有遮挡。
> > [image]
> 
> 非常感谢您的回复，我修改了相关参数，但是效果好像和您的有较大差异

另一个建议：在odom不漂的情况下, LI-Init的参数gyr_cov越大越好，越大时差标定越准确（但是太大odom容易漂）。我改成5之后，重新标定的时间差是-0.01, FAST-LIO2的参数如下：
[image]
[image]
效果如下（基本回到原点）
[image]

af-doom, 2022-11-04:
> 

真的太感谢您这么快的回复，我得到了好的结果，您说的在odom不漂的情况下, LI-Init的参数gyr_cov越大越好，越大时差标定越准确，这个原理是啥呢，因为这个参数给的默认值才是0.5，那么针对不同的雷达惯导，应该如何从原理出发调参数呢?麻烦您了

zfc-zfc, 2022-11-04:
> > 
> 
> 真的太感谢您这么快的回复，我得到了好的结果，您说的在odom不漂的情况下, LI-Init的参数gyr_cov越大越好，越大时差标定越准确，这个原理是啥呢，因为这个参数给的默认值才是0.5，那么针对不同的雷达惯导，应该如何从原理出发调参数呢?麻烦您了

这个原理大致是因为 EKF这个滤波器本身有时间延迟，gyr_cov是角速度的过程噪声，越大表示越相信观测，那么时延就会越低，从而时差也越接近真实。但是如果过程噪声太大，很容易导致odom抖动得很厉害，不平滑，从而漂掉。
一般来说，针对livox这种非重复式扫描的，他的FoV不受分帧影响，雷达里程计更鲁棒，所以livox的雷达 填个50一般都没问题。

对于机械雷达，线数越高，点越多，里程计也更鲁棒，但是其FoV会随着分帧数增大而减小，鲁棒性不如非重复式扫描。
128线的，填5~10一般都OK，16线就很脆弱了，填小一点里程计不容易飞，但是时差没那么准确。如果你给予激励的手法比较好，雷达运动比较匀速，那么给大一点得到更准的时差。

af-doom, 2022-11-04:
> 这个原理大致是因为 EKF这个滤波器本身有时间延迟，gyr_cov是角速度的过程噪声，越大表示越相信观测，那么时延就会越低，从而时差也越接近真实。但是如果过程噪声太大，很容易导致odom抖动得很厉害，不平滑，从而漂掉。 一般来说，针对livox这种非重复式扫描的，他的FoV不受分帧影响，雷达里程计更鲁棒，所以livox的雷达 填个50一般都没问题。 [...]

好的，非常感谢您如此快速的回复。

closed 2022-11-07 by zfc-zfc