hku-mars/LiDAR_IMU_Init #40: ceres error

Ahmad-Drak, 2022-11-04:
i get the following error when building from source, it is related to ceres. i couldnt find a reason or a solution for it. Any suggestions?
```
93%] Built target normal_filter_comparison_demo
[ 94%] Built target octomap_to_gridmap_demo
[ 97%] Built target filters_demo
[ 97%] Built target grid_map_to_image_demo
/home/ahmad/workspace/catkin_ws/src/LiDAR_IMU_Init/include/LI_init/LI_init.cpp: In member function ‘void LI_Init::solve_Rotation_only()’:
/home/ahmad/workspace/catkin_ws/src/LiDAR_IMU_Init/include/LI_init/LI_init.cpp:324:51: error: expected type-specifier
  324 |     ceres::LocalParameterization *quatParam = new ceres::QuaternionParameterization();
      |                                                   ^~~~~
/home/ahmad/workspace/catkin_ws/src/LiDAR_IMU_Init/include/LI_init/LI_init.cpp:326:58: error: no matching function for call to ‘ceres::Problem::AddParameterBlock(double [4], int, ceres::LocalParameterization*&)’
  326 |     problem_rot.AddParameterBlock(R_LI_quat, 4, quatParam);
      |                                                          ^
In file included from /usr/local/include/ceres/ceres.h:64,
                 from /home/ahmad/workspace/catkin_ws/src/LiDAR_IMU_Init/include/LI_init/LI_init.h:15,
                 from /home/ahmad/workspace/catkin_ws/src/LiDAR_IMU_Init/include/LI_init/LI_init.cpp:1:
/usr/local/include/ceres/problem.h:261:8: note: candidate: ‘void ceres::Problem::AddParameterBlock(double*, int)’
  261 |   void AddParameterBlock(double* values, int size);
      |        ^~~~~~~~~~~~~~~~~
/usr/local/include/ceres/problem.h:261:8: note:   candidate expects 2 arguments, 3 provided
/usr/local/include/ceres/problem.h:274:8: note: candidate: ‘void ceres::Problem::AddParameterBlock(double*, int, ceres::Manifold*)’
  274 |   void AddParameterBlock(double* values, int size, Manifold* manifold);
      |        ^~~~~~~~~~~~~~~~~
/usr/local/include/ceres/problem.h:274:62: note:   no known conversion for argument 3 from ‘ceres::LocalParameterization*’ to ‘ceres::Manifold*’
  274 |   void AddParameterBlock(double* values, int size, Manifold* manifold);
      |                                                    ~~~~~~~~~~^~~~~~~~
/home/ahmad/workspace/catkin_ws/src/LiDAR_IMU_Init/include/LI_init/LI_init.cpp: In member function ‘void LI_Init::solve_Rot_bias_gyro(double&)’:
/home/ahmad/workspace/catkin_ws/src/LiDAR_IMU_Init/include/LI_init/LI_init.cpp:360:51: error: expected type-specifier
  360 |     ceres::LocalParameterization *quatParam = new ceres::QuaternionParameterization();
      |                                                   ^~~~~
/home/ahmad/workspace/catkin_ws/src/LiDAR_IMU_Init/include/LI_init/LI_init.cpp:363:62: error: no matching function for call to ‘ceres::Problem::AddParameterBlock(double [4], int, ceres::LocalParameterization*&)’
  363 |     problem_ang_vel.AddParameterBlock(R_LI_quat, 4, quatParam);
      |                                                              ^
In file included from /usr/local/include/ceres/ceres.h:64,
                 from /home/ahmad/workspace/catkin_ws/src/LiDAR_IMU_Init/include/LI_init/LI_init.h:15,
                 from /home/ahmad/workspace/catkin_ws/src/LiDAR_IMU_Init/include/LI_init/LI_init.cpp:1:
/usr/local/include/ceres/problem.h:261:8: note: candidate: ‘void ceres::Problem::AddParameterBlock(double*, int)’
  261 |   void AddParameterBlock(double* values, int size);
      |        ^~~~~~~~~~~~~~~~~
/usr/local/include/ceres/problem.h:261:8: note:   candidate expects 2 arguments, 3 provided
/usr/local/include/ceres/problem.h:274:8: note: candidate: ‘void ceres::Problem::AddParameterBlock(double*, int, ceres::Manifold*)’
  274 |   void AddParameterBlock(double* values, int size, Manifold* manifold);
      |        ^~~~~~~~~~~~~~~~~
/usr/local/include/ceres/problem.h:274:62: note:   no known conversion for argument 3 from ‘ceres::LocalParameterization*’ to ‘ceres::Manifold*’
  274 |   void AddParameterBlock(double* values, int size, Manifold* manifold);
      |                                                    ~~~~~~~~~~^~~~~~~~
/home/ahmad/workspace/catkin_ws/src/LiDAR_IMU_Init/include/LI_init/LI_init.cpp: In member function ‘void LI_Init::solve_trans_biasacc_grav()’:
/home/ahmad/workspace/catkin_ws/src/LiDAR_IMU_Init/include/LI_init/LI_init.cpp:424:51: error: expected type-specifier
  424 |     ceres::LocalParameterization *quatParam = new ceres::QuaternionParameterization();
      |                                                   ^~~~~
/home/ahmad/workspace/catkin_ws/src/LiDAR_IMU_Init/include/LI_init/LI_init.cpp:427:59: error: no matching function for call to ‘ceres::Problem::AddParameterBlock(double [4], int, ceres::LocalParameterization*&)’
  427 |     problem_acc.AddParameterBlock(R_GL0_quat, 4, quatParam);
      |                                                           ^
In file included from /usr/local/include/ceres/ceres.h:64,
                 from /home/ahmad/workspace/catkin_ws/src/LiDAR_IMU_Init/include/LI_init/LI_init.h:15,
                 from /home/ahmad/workspace/catkin_ws/src/LiDAR_IMU_Init/include/LI_init/LI_init.cpp:1:
/usr/local/include/ceres/problem.h:261:8: note: candidate: ‘void ceres::Problem::AddParameterBlock(double*, int)’
  261 |   void AddParameterBlock(double* values, int size);
      |        ^~~~~~~~~~~~~~~~~
/usr/local/include/ceres/problem.h:261:8: note:   candidate expects 2 arguments, 3 provided
/usr/local/include/ceres/problem.h:274:8: note: candidate: ‘void ceres::Problem::AddParameterBlock(double*, int, ceres::Manifold*)’
  274 |   void AddParameterBlock(double* values, int size, Manifold* manifold);
      |        ^~~~~~~~~~~~~~~~~
/usr/local/include/ceres/problem.h:274:62: note:   no known conversion for argument 3 from ‘ceres::LocalParameterization*’ to ‘ceres::Manifold*’
  274 |   void AddParameterBlock(double* values, int size, Manifold* manifold);
      |                                                    ~~~~~~~~~~^~~~~~~~
make[2]: *** [LiDAR_IMU_Init/CMakeFiles/li_init.dir/build.make:89: LiDAR_IMU_Init/CMakeFiles/li_init.dir/include/LI_init/LI_init.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:9866: LiDAR_IMU_Init/CMakeFiles/li_init.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make" failed
```

Taeyoung96, 2022-11-06:
@Ahmad-Drak   
Which ceres version did you use?  
I'm using the ceres-solver 2.0.0 version and there was no build problem.

Ahmad-Drak, 2022-11-07:
@Taeyoung96 I am using ceres-solver-2.1.0 as this version is needed for another package im using.
When opening the link indicated in the readme  [here](https://github.com/hku-mars/LiDAR_IMU_Init#14-ceres-solver) it shows ceres 2.1

Ahmad-Drak, 2022-11-07:
looking at the version history [here](http://ceres-solver.org/version_history.html) it seems that version 2.1 is not backward compatiable. Specifically:
`[LocalParameterization](http://ceres-solver.org/nnls_modeling.html#_CPPv4N5ceres21LocalParameterizationE) is deprecated. It will be removed in version 2.2. Use [Manifold](http://ceres-solver.org/nnls_modeling.html#_CPPv4N5ceres8ManifoldE) instead.`

i assume the solution is to either use ceres version 2.0 or change the `solve_Rotation_only()` function in LI_Init.cpp as suggested in the version history.

Ahmad-Drak, 2022-11-07:
If using ceres-solver > 2.0 (im using 2.1) you can apply this fix to make it compile:
in  [LI_Init.cpp](https://github.com/hku-mars/LiDAR_IMU_Init/blob/main/include/LI_init/LI_init.cpp) change 3 lines (324,360,424)

    ceres::LocalParameterization *quatParam = new ceres::QuaternionParameterization();

to

`ceres::Manifold *quatParam = new ceres::QuaternionManifold();`

closed 2022-11-07 by Ahmad-Drak