hku-mars/LiDAR_IMU_Init #41: Question about Extrinsic Translation algorithm.

Taeyoung96, 2022-11-06:
@zfc-zfc   
Thanks for your great work.  
I have a question because the code is different from equation (23) in the paper.  

The equation (23) shown in the paper is as follows.  

<p align="center">[image]</p>  

And I understood the code that implemented this as `LI_Init::solve_trans_biasacc_grav()`.  
Looking at the residual function `Linear_acc_Cost`, I find that the variable `R_LL0_T`(the estimated attitude (rotation matrix) at the end lidar point), which is not mentioned in the paper, continues to be used.  

Why does this variable continue to be used?  
I'd appreciate it if you could give me an answer.

zfc-zfc, 2022-11-07:
> 

Please refer to Equation (22)

> bias_aL

[image]

Taeyoung96, 2022-11-07:
@zfc-zfc   
Thank you for your quick response. :+1:

closed 2022-11-07 by Taeyoung96