hku-mars/LiDAR_IMU_Init #42: "Leaf size is too small for the input dataset"   when "downSizeFilterSurf.filter(*feats_down_body)"

skyengine83, 2022-11-18:
run to
`downSizeFilterSurf.filter(*feats_down_body);`
warn 
`[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow.`

zfc-zfc, 2022-11-19:
> Is there wrong with Lidar data? run to `downSizeFilterSurf.filter(*feats_down_body);` warn `[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow.`

filter_size_surf (meter): It is recommended that filter_size_surf = 0.05~0.25 for indoor scenes, filter_size_surf = 0.5 for outdoor scenes.
filter_size_map (meter): It is recommended that filter_size_map = 0.15~0.35 for indoor scenes, filter_size_map = 0.5 for outdoor scenes.

If this warning still exists, try to increase these two parameters.

zfc-zfc, 2022-11-19:
> Is there wrong with Lidar data? run to `downSizeFilterSurf.filter(*feats_down_body);` warn `[pcl::VoxelGrid::applyFilter] Leaf size is too small for the input dataset. Integer indices would overflow.`

filter_size_surf (meter): It is recommended that filter_size_surf = 0.050.25 for indoor scenes, filter_size_surf = 0.5 for outdoor scenes.
filter_size_map (meter): It is recommended that filter_size_map = 0.150.35 for indoor scenes, filter_size_map = 0.5 for outdoor scenes.
If this warning still exists, try to increase these two parameters.

closed 2022-11-26 by zfc-zfc