hku-mars/LiDAR_IMU_Init #44: [laserMapping-1] process has finished cleanly

ZeeshanshariffNS, 2022-11-25:
We are using a ouster lidar and voxl-imu to run LIO and to estimate extrinsic parameters we are using   "roslaunch lidar_imu_init ouster.launch"  but it terminates with the message "process has finished cleanly"

basically the Lidar and IMU buffer is zero in laserMapping node before this termination occurs.
Can you please let us the cause of the error and the way to solve this problem.

LOGS AS DISPLAYED ON THE  TERMINAL 

root@qrb5165-rb5:~/ws_livox# roslaunch lidar_imu_init ouster.launch 
... logging to /root/.ros/log/15dc4692-6cbb-11ed-9c1c-000ec6817901/roslaunch-qrb5165-rb5-29011.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://qrb5165-rb5:41621/

SUMMARY
========

PARAMETERS
 * /common/imu_topic: /ouster/imu
 * /common/lid_topic: /ouster/points
 * /cube_side_length: 2000.0
 * /initialization/Rot_LI_cov: [5e-05, 5e-05, 5e...
 * /initialization/Trans_LI_cov: [0.0001, 0.0001, ...
 * /initialization/cut_frame: True
 * /initialization/cut_frame_num: 3
 * /initialization/data_accum_length: 300
 * /initialization/mean_acc_norm: 9.805
 * /initialization/online_refine_time: 15
 * /initialization/orig_odom_freq: 10
 * /mapping/acc_cov: 2
 * /mapping/b_acc_cov: 0.0001
 * /mapping/b_gyr_cov: 0.0001
 * /mapping/det_range: 150.0
 * /mapping/filter_size_map: 0.15
 * /mapping/filter_size_surf: 0.05
 * /mapping/gyr_cov: 50
 * /max_iteration: 5
 * /pcd_save/interval: -1
 * /pcd_save/pcd_save_en: False
 * /point_filter_num: 1
 * /preprocess/blind: 1
 * /preprocess/feature_extract_en: False
 * /preprocess/lidar_type: 3
 * /preprocess/scan_line: 32
 * /publish/dense_publish_en: True
 * /publish/path_en: True
 * /publish/scan_bodyframe_pub_en: True
 * /publish/scan_publish_en: True
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    laserMapping (lidar_imu_init/li_init)

ROS_MASTER_URI=http://localhost:11311

process[laserMapping-1]: started with pid [29160]
[laserMapping-1] process has finished cleanly
log file: /root/.ros/log/15dc4692-6cbb-11ed-9c1c-000ec6817901/laserMapping-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
root@qrb5165-rb5:~/ws_livox# roslaunch lidar_imu_init ouster.launch 
... logging to /root/.ros/log/15dc4692-6cbb-11ed-9c1c-000ec6817901/roslaunch-qrb5165-rb5-20158.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://qrb5165-rb5:35381/

SUMMARY
========

PARAMETERS
 * /common/imu_topic: /ouster/imu
 * /common/lid_topic: /ouster/points
 * /cube_side_length: 2000.0
 * /initialization/Rot_LI_cov: [5e-05, 5e-05, 5e...
 * /initialization/Trans_LI_cov: [0.0001, 0.0001, ...
 * /initialization/cut_frame: True
 * /initialization/cut_frame_num: 3
 * /initialization/data_accum_length: 300
 * /initialization/mean_acc_norm: 9.805
 * /initialization/online_refine_time: 15
 * /initialization/orig_odom_freq: 10
 * /mapping/acc_cov: 2
 * /mapping/b_acc_cov: 0.0001
 * /mapping/b_gyr_cov: 0.0001
 * /mapping/det_range: 150.0
 * /mapping/filter_size_map: 0.15
 * /mapping/filter_size_surf: 0.05
 * /mapping/gyr_cov: 50
 * /max_iteration: 5
 * /pcd_save/interval: -1
 * /pcd_save/pcd_save_en: False
 * /point_filter_num: 1
 * /preprocess/blind: 1
 * /preprocess/feature_extract_en: False
 * /preprocess/lidar_type: 3
 * /preprocess/scan_line: 32
 * /publish/dense_publish_en: True
 * /publish/path_en: True
 * /publish/scan_bodyframe_pub_en: True
 * /publish/scan_publish_en: True
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    laserMapping (lidar_imu_init/li_init)

ROS_MASTER_URI=http://localhost:11311

process[laserMapping-1]: started with pid [20268]
Multi thread started 
lidar_type: 3
LiDAR-only odometry starts.
~~~~/root/ws_livox/src/LiDAR_IMU_Init/ file opened

[Exit]: Exit the process.
[WARN]: Online refinement not finished yet.
[Final Result] Rotation LiDAR to IMU    =  0.000000 -0.000000  0.000000 deg
[Final Result] Translation LiDAR to IMU = 0.000000 0.000000 0.000000 m
[Final Result] Time Lag IMU to LiDAR    = 0.00000000 s 
[Final Result] Bias of Gyroscope        = 0.000000 0.000000 0.000000 rad/s
[Final Result] Bias of Accelerometer    = 0.000000 0.000000 0.000000 m/s^2
[Final Result] Gravity in World Frame   = 0.000000 0.000000 0.000000 m/s^2
Rebuild thread terminated normally
[laserMapping-1] process has finished cleanly
log file: /root/.ros/log/15dc4692-6cbb-11ed-9c1c-000ec6817901/laserMapping-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete

Bala-Ramachandran-Joulea, 2023-02-17:
Hey, Did you solve this issue?

edwardes2201, 2023-02-21:
hello, I also have the same error, I'm using ubuntu 20.04 installed on a pi4 and I can't run. however on a normal pc with ubuntu 18.04 it runs normally without error. If anyone knows how to fix it, it would be a great help. greetings

closed 2023-07-12 by zfc-zfc