hku-mars/LiDAR_IMU_Init #48: LaserMapping-cpp:1283:1: ERROR

nes9966, 2023-01-08:
Hello.  I have the following error when compiling Lidar-IMU-Init.
I am running 11th Gen Intel(R) Core(TM) i7-11800H @ 2.30GHz   2.30 GHz.  Virtual Box - Ubuntu 18.04
[image]

I have tried to apply the fix on Issue #7
But then I get this error below.  Not sure if I applied the fix correctly since I am not a coder.
[image]
 
Can some one help with this? Thank you in advance.

nes9966, 2023-01-22:
My apologies.  I have been trying to figure this error but I cannot.  I am not a coder, so I am not sure how to proceed.  I am running a Livox Mid40.
Would I need to install the program first, then change to my specific parameters?
or do I adjust parameters before I install?  
I have tried to trace the expected '}' at the end of input.  but all brackets are closed. no open.  Does anyone have a clue?

Happy New Year!

closed 2023-02-06 by nes9966

esolk, 2023-04-04:
I am anxious to know how you resolve this error. I got the same error on my Jetson Xavier NX, which is going well on my laptop.
 

> Thank you, Tomato1107 for helping me resolve this issue.

nes9966, 2023-04-06:
Hello.  Sorry for the late reply.  I started this on a TX1 but could not
get it to work.  Went to my laptop and it worked after sorting out
libraries.  It may be that the version of the installed libraries are
different from those in github.
You can try ubuntu20.04 noetic, it's easy to set up pcl, Eigen, ceres and
opencv with the following commands:
sudo apt install libpcl-dev libopencv-dev libeigen3-dev libceres-dev
In this way, you don't need to manually install many libraries, and you can
also avoid compilation problems caused by inconsistencies in libraries.
Hope this helps.