hku-mars/LiDAR_IMU_Init #50: Build Error

jin-bae-joulea, 2023-02-20:
I'm using ROS Noetic and catkin tools. When I clone the repository and run catkin build, it gives me a compilation error saying that I'm missing a curly bracket at line 1251, saying it expected '}' at the end of input:

```
..../src/LiDAR_IMU_Init/src/laserMapping.cpp:1251:1: error: expected ‘}’ at end of input 1251 | }
```
How can I go about fixing this issue? Thanks.

edwardes2201, 2023-02-21:
Hello, try with ubuntu 18.04 there it runs without problems. In 20.04 it is not possible because of what you say.

zfc-zfc, 2023-02-21:
> I'm using ROS Noetic and catkin tools. When I clone the repository and run catkin build, it gives me a compilation error saying that I'm missing a curly bracket at line 1251, saying it expected '}' [...]

run "catkin_make -j"

edwardes2201, 2023-02-22:
n file included from /usr/include/pcl-1.10/pcl/point_types.h:42,
                 from /home/ubuntu/lidarin/catkin_ws/src/LiDAR_IMU_Init/include/common_lib.h:6,
                 from /home/ubuntu/lidarin/catkin_ws/src/LiDAR_IMU_Init/src/IMU_Processing.hpp:14,
                 from /home/ubuntu/lidarin/catkin_ws/src/LiDAR_IMU_Init/src/laserMapping.cpp:36:
/usr/include/pcl-1.10/pcl/pcl_macros.h:94:9: note: declared here
   94 |   using uint16_t [[deprecated("use std::uint16_t instead of pcl::uint16_t")]] = std::uint16_t;
      |         ^~~~~~~~
/home/ubuntu/lidarin/catkin_ws/src/LiDAR_IMU_Init/src/laserMapping.cpp: In function ‘int main(int, char**)’:
/home/ubuntu/lidarin/catkin_ws/src/LiDAR_IMU_Init/src/laserMapping.cpp:1254:1: error: expected ‘}’ at end of input
 1254 | }
      | ^
/home/ubuntu/lidarin/catkin_ws/src/LiDAR_IMU_Init/src/laserMapping.cpp:766:33: note: to match this ‘{’
  766 | int main(int argc, char **argv) {
      |                                 ^
make[2]: *** [LiDAR_IMU_Init/CMakeFiles/li_init.dir/build.make:63: LiDAR_IMU_Init/CMakeFiles/li_init.dir/src/laserMapping.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:638: LiDAR_IMU_Init/CMakeFiles/li_init.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j" failed
ubuntu@ubuntu:~/lidarin/catkin_ws$ 

hello, with catkin_make -j the error still persists.

ki4ma20, 2023-03-04:
Same issue is occurred on building in my aarch64 architecture  environment. I tried to seek the codes and found a solution.

This issue occurs when the macro "MP_EN" is not defined.
At the [line 967](https://github.com/hku-mars/LiDAR_IMU_Init/blob/8ce7918e1053c3f6aa2afdcfca7474561de12910/src/laserMapping.cpp#L967) of laserMapping.cpp, #ifdef MP_EN is described and the corresponded #endif is described at the [line 1138](https://github.com/hku-mars/LiDAR_IMU_Init/blob/8ce7918e1053c3f6aa2afdcfca7474561de12910/src/laserMapping.cpp#L1138).

Since the bracket { of the [line 970](https://github.com/hku-mars/LiDAR_IMU_Init/blob/8ce7918e1053c3f6aa2afdcfca7474561de12910/src/laserMapping.cpp#L970) are closed at the [line 1136](https://github.com/hku-mars/LiDAR_IMU_Init/blob/8ce7918e1053c3f6aa2afdcfca7474561de12910/src/laserMapping.cpp#L1136) , the bracket remains as opened when MP_EN is not defined.

The macro MP_EN seems to be used to switch to use the "omp parallel for" of OpenMP according to the architecture.
And I think that the #endif in the [line 1138](https://github.com/hku-mars/LiDAR_IMU_Init/blob/8ce7918e1053c3f6aa2afdcfca7474561de12910/src/laserMapping.cpp#L1138) can be moved to before the [line 970](https://github.com/hku-mars/LiDAR_IMU_Init/blob/8ce7918e1053c3f6aa2afdcfca7474561de12910/src/laserMapping.cpp#L970) without changing the meaning.

I attach the my zip-compressed code which can be build, Please check the difference.
[laserMapping.zip](https://github.com/hku-mars/LiDAR_IMU_Init/files/10888308/laserMapping.zip)

zfc-zfc, 2023-04-20:
> 

Thank you for your correction. I will update the code as soon as possible.

closed 2023-07-12 by zfc-zfc