hku-mars/LiDAR_IMU_Init #51: horizon标定地图混乱

gongjun136, 2023-02-23:
作者您好，感谢您的工作，我运行的horizon.launch，但是标定的地图构建很乱，请问下这个框架是否适用horizon

KnineF, 2023-02-24:
[image]
[image]
[image]

作者您好，用horizon雷达和原版驱动，运行livox_horizon.launch，在初始化的三轴旋转时正常，得到图一的地图，到后面online refinement过程中，就出现图二到图三的情况，坐标轴一直往远处飘。我已检查过原版驱动发布的imu话题是弧度制，所以不太清楚原因

zfc-zfc, 2023-03-22:
> 

场景太小，建议到开阔的室外或者大一点的房间里测试标定。另外要保持激励充分。