hku-mars/LiDAR_IMU_Init #61: it takes a long time for the progress bar to appear during initialization

wangyicxy, 2023-04-06:
I use the velodyne 32 line lidar to calibrate with the imu, but it takes a long time for the progress bar to appear during initialization, and it always gets stuck, and sometimes the program will suddenly terminate. What is the reason?

qimao7213, 2023-04-10:
你好，我也遇到了类似的问题，请问你解决了吗？我发现相比于实时校正，可以先将数据录成bag包，然后再跑程序，效果要好一些，也可以调试cong里的一也可以调试cong里的一些参数。些参数。

wangyicxy, 2023-04-12:
> 你好，我也遇到了类似的问题，请问你解决了吗？我发现相比于实时校正，可以先将数据录成bag包，然后再跑程序，效果要好一些，也可以调试cong里的一也可以调试cong里的一些参数。些参数。

我解决了，我发现在初始化的时候需要将雷达和惯导进行一些移动，会缩短初始化的时间，但是标定结果每次的差别都比较大

qimao7213, 2023-04-12:
你好，谢谢你的回复。我们在使用VLP32的时候，将配置文件中的frame_cut什么的参数设置为了1（如果太大进度条就直接满了，然后飘了），将date_length_accumulation什么的参数从800改成了400，得到了一个比较合理的结果，但是还需要后续的实验判断这个结果是不是可以提升建图的精度。

wangyicxy, 2023-06-15:
> > 你好，我也遇到了类似的问题，请问你解决了吗？我发现相比于实时校正，可以先将数据录成bag包，然后再跑程序，效果要好一些，也可以调试cong里的一也可以调试cong里的一些参数。些参数。
> 
> 你好，请问怎么跑已经录制好的.bag数据，我在项目中没看到数据应该放在什么路径中

您好，直接rosbag play播包即可

closed 2023-07-12 by zfc-zfc