hku-mars/LiDAR_IMU_Init #63: 关于最后联合优化时候IMU的bias值依然不稳定的问题

jiejie567, 2023-04-19:
hi，非常棒的工作。我这边尝试用mid-360及内置的IMU运行LiDAR_IMU_init及原版的Fast-lio，我的目的只是想看下加速度计的bias。两者的参数都设置为一致，两者都顺利运行，点云也清晰没有重影，最后两者建的图也基本一致。但是，我发现，同一个数据集，两个框架计算的bias值是不同的，甚至正负号都不一致，而且在优化的时候会发散，请问这是什么原因？是iekf框架导致的么？再次感谢你们开源如此优秀的作品。

zfc-zfc, 2023-04-20:
> 

请问FAST-LIO和LI-Init各自的bias结果是多少呢？因为这两个框架所优化的变量和约束类型差异还是比较大的，所以结果有差异很正常，我们的目的是能让任何lidar imu设备能成功运行LIO，只要能完成这个目标也就算完成使命了。另外“优化的时候会发散”是指哪个框架在优化哪个状态时的现象呢？

jiejie567, 2023-04-21:
感谢 @zfc-zfc 的快速回复！优化会发散可能是我不准确的描述，下图是用你们提供的包运行的结果，最后的bias好像还没收敛。
[image]
下图是用fast_lio用一样的配置参数运行的，直接设置了时间差-0.1s，   acc_cov: 50  gyr_cov: 2
[image]
从图中可以看到两者的bias不一致，有些正负好像也不同。
另外想问下设置为acc_cov， gyr_cov设置那么大的原因？

zfc-zfc, 2023-04-21:
> 

acc_cov和gyr_cov在两个框架下的意义是不一样的；在FAST-LIO框架下，acc_cov是加速度计的噪声参数，gyr_cov是陀螺仪的噪声参数；但是在LI-Init框架下，acc_cov是用来驱动线速度的白噪声（EKF的过程噪声），而gyr_cov是用来驱动角速度的白噪声（过程噪声）。因为意义不同，所以参数取值也应当有区别。至于在LI-Init中gyr_cov设置很大的原因可以参考 #39。

在使用FAST-LIO时，建议仅仅将外参和时差填入，gyr_cov, acc_cov使用原本的0.1.

jiejie567, 2023-04-22:
谢谢，看了你在 #39 中的答复，收获颇多，基本上明白了。还有一个小疑问是，我重新运行了，但是结果和上次差的不太多，两个估计的各个轴的bias正负有些不同，这大概是什么原因产生的呢？

closed 2023-07-12 by zfc-zfc