hku-mars/LiDAR_IMU_Init #65: 标定出来的平移不准确

bruce87en, 2023-04-25:
雷达是禾赛的pandarXT-16，在一个比较有结构的地方采集数据。
标定出来的旋转非常准确，且比较稳定：
Rotation LiDAR to IMU    = -64.505412  -0.040517   3.325081 deg
但标出来的平移非常不准备，rosbag play时，使用-s跳过数据，跳过的时间不同会导致标定出来的平移有非常大的差异。
-s 10时：
[Final Result] Translation LiDAR to IMU = -0.112968 -0.067292  0.143370 m
-s 12时：
[Final Result] Translation LiDAR to IMU = 0.308087 0.575553 1.322567 m
而使用-s跳过不同的时间，对旋转是没有影响的，旋转如上面所示，非常稳定。
RVIZ中显示的地图与轨迹看起来都非常OK。
请问题这可能是什么原因导致的？

zfc-zfc, 2023-04-27:
平移的标定是靠align 雷达里程计的加速度和IMU的加速度计来获得的，而雷达里程计的加速度是靠线速度差分出来的，相比于旋转的标定使用角速度，差分得到的线加速度噪声会更大；同时文章的公式22和23也可以看出标定平移时需要优化的变量很多，如果线加速度的激励不够会导致错误的收敛。根据你的描述来看，应该整个过程的加速度激励不够大，这边建议运行一下matlab_code文件夹下的result_plot.m，看看曲线贴合的情况。如果激励足够，无论你跳过多长时间，结果应该是差不多的，就像下图这样
[image]

zfc-zfc, 2023-04-27:
> 雷达是禾赛的pandarXT-16，在一个比较有结构的地方采集数据。 标定出来的旋转非常准确，且比较稳定： Rotation LiDAR to IMU = -64.505412 -0.040517 3.325081 deg 但标出来的平移非常不准备，rosbag play时，使用-s跳过数据，跳过的时间不同会导致标定出来的平移有非常大的差异。 -s 10时： [Final Result] [...]

另外，建议在设备静置时开始采集数据；-s 10可能设备正在运动过程中，这会导致ikd-tree的初始地图比较混乱，会影响结果。

bruce87en, 2023-04-28:
感谢回复，跑了一下matlab脚本，的确是加速度激励不够。

closed 2023-04-28 by bruce87en