hku-mars/LiDAR_IMU_Init #66: Failed to calibrate using the sample data Hesai_library.bag

CharlieV5, 2023-05-07:
I download the sample data here: [Lidar IMU Initialization Datasets](https://connecthkuhk-my.sharepoint.com/:f:/g/personal/zhufc_connect_hku_hk/EgdJ_F763sVOnkUNBRv-op8BmNL7eZrxETu2zSEAoiRX4A?e=cbNiJI).
I have test the Hesai_library.bag and Hesai_apartment1.bag. The Initialization has never reach 100% when the rosbag play ends, as the screen shot below.
[image]
[image]

zfc-zfc, 2023-05-11:
Adjusting data_accum_length and cut_frame_num would fix this, please try setting data_accum_length to 300 and cut_frame_num to 4.

CharlieV5, 2023-05-13:
> Adjusting data_accum_length and cut_frame_num would fix this, please try setting data_accum_length to 300 and cut_frame_num to 4.

Thank you!

closed 2023-05-13 by CharlieV5