hku-mars/LiDAR_IMU_Init #67: 适用于地面平台吗

Amnesiacers, 2023-05-17:
你好，很棒的工作！
我试着在地面平台上进行标定，但发现x轴，y轴的激励增长异常缓慢（很长时间后仍低于10%），我试着选了如图的一个坡，在上面绕圈，但仍然很慢，请问这个标定工作如何在地面平台pitch和roll都限制比较大的情况下有效呢
[image]
[image]

zfc-zfc, 2023-05-21:
> 你好，很棒的工作！ 我试着在地面平台上进行标定，但发现x轴，y轴的激励增长异常缓慢（很长时间后仍低于10%），我试着选了如图的一个坡，在上面绕圈，但仍然很慢，请问这个标定工作如何在地面平台pitch和roll都限制比较大的情况下有效呢 [image] [image]

你好，由于IMU需要在各个维度都有充分激励才能获得全维度的标定结果，所以目前这个工具适合类似手持这样的可全维度运动的平台，地面平台不适用。

Amnesiacers, 2023-05-22:
非常感谢解答，我试着降低x、y维度激励程度的阈值，但标定结果不尽如人意，在目前这种地面平台确实无法满足各个维度都得到充分激励的情况下，对其进行标定是无解的吗，或者目前有其他比较适用的方法吗

wwtinwhu, 2023-05-23:
> 非常感谢解答，我试着降低x、y维度激励程度的阈值，但标定结果不尽如人意，在目前这种地面平台确实无法满足各个维度都得到充分激励的情况下，对其进行标定是无解的吗，或者目前有其他比较适用的方法吗

你好，我也有做相关方法；之前也都是测试的运动激励充分的数据。这个数据方便发我试一下吗？

zfc-zfc, 2023-05-23:
> 

https://github.com/APRIL-ZJU/OA-LICalib 也许可以试试这个方法，文章里说小车平台也可以标定。

yeyanan93, 2023-05-26:
> > 
> 
> https://github.com/APRIL-ZJU/OA-LICalib 也许可以试试这个方法，文章里说小车平台也可以标定。

请问后续有继续研究车载平台的lidar_imu标定吗

closed 2023-07-12 by zfc-zfc