hku-mars/LiDAR_IMU_Init #68: 雷达和imu标定

huangwefeng121, 2023-06-07:
首先感谢大佬将这么优秀的项目开源。我在此有点疑问，我想用速腾的M1激光雷达（视野范围正前方120度）和外部的imu传感器（轮趣科技的）做两者的外参标定，能实现吗？还是说必须用livox的雷达和内部自带的imu才能实现外参标定

zfc-zfc, 2023-06-07:
> 雷达

按理是可以的，不过可能需要对该种雷达的点云数据做一些适配，比如在common_lib.h line55添加一种新的雷达类型，然后增加相应的回调函数和预处理函数。

huangwefeng121, 2023-06-08:
> > 雷达
> 
> 按理是可以的，不过可能需要对该种雷达的点云数据做一些适配，比如在common_lib.h line55添加一种新的雷达类型，然后增加相应的回调函数和预处理函数

那请问我能先录制好lidar和imu的.bag数据，然后再用您提供的工具箱计算标定参数吗？录制的时候按照您说的，先静止5s，然后再三轴方向给充分的激励。如果可以，我需要将录制的.bag数据放在什么路径下读取，因为我暂时在您的工具包中没看到这部分

huangwefeng121, 2023-06-08:
> 有试过那个lidar_align吗，不过如果你做到了，请联系我。谢谢

目前还在尝试中，请问您那边有什么进展吗？我们可以一起交流一下

huangwefeng121, 2023-06-09:
[image]
谢谢作者的指导，目前已经把我雷达的数据跑起来了，但是我发现两个问题希望有时间能解惑一下
问题一：我这imu轨迹偏移较大，可以通过调整什么参数来优化吗？或者我转动的时候角速度慢点，是否能够有所改善？
问题二：点云跑完之后，pointCloud2显示红色报错：For frame [aglidar]:Frame [aglidar] dose not exit.   我的lidar的topic是aglidar_points

zfc-zfc, 2023-06-09:
> 错
这种情况我比较怀疑是每个点的时间戳错误，一定要确保代码里得到了每个点相对于帧头正确的偏移时间。

huangwefeng121, 2023-06-12:
我的数据中每个点云都有自己的时间戳（系统时间），但是每帧里面每100个点对应的时间戳是一样的，并不能做到每个点的时间不同，这应该没问题吧

huangwefeng121, 2023-06-16:
[image]
博主，我还发现一个问题，我的lidar和imu全程没有静止，但是最后出来的轨迹还是有一个偏移，这是什么缘故

huangwefeng121, 2023-06-19:
> 从图里看，周围地图都是糊的，说明lidar odometry是漂的，所以什么样的轨迹都可能，反正都是错的。至于为什么odometry会漂，建议检查单点时间戳的单位（代码在preprocess.cpp 233~297），imu角速度的单位是否正确。你也可以上传你的数据到onedrive，我帮你看看原因。

电脑原因无法上传OneDrive，我上传了百度网盘，麻烦作者可以帮忙看看数据问题
链接：https://pan.baidu.com/s/12IjChpo09n_THjBbvAdL-w?pwd=8aad 
提取码：8aad
我在preprocess.cpp中输出262行的msg->header.stamp，一直显示0，这会有什么问题吗

huangwefeng121, 2023-06-20:
[image]

这是跑出来的激励，好像对的不怎么准确

huangwefeng121, 2023-06-25:
你好，我的数据有什么原因吗，您那边能不能帮忙定位一下，因为用的不是工程里推荐的几款雷达，所以担心有什么点云底层的问题

zfc-zfc, 2023-06-30:
> 你好，我的数据有什么原因吗，您那边能不能帮忙定位一下，因为用的不是工程里推荐的几款雷达，所以担心有什么点云底层的问题

我看了一下场景，障碍物太近了，雷达近处精度很差的，需要拿到开阔一点的地方
[image]

huangwefeng121, 2023-07-03:
这个当时场景前方5m左右有墙面。您那边分析我的点云信息是否正确，缺少ring这个参数有影响吗？我的数据中每个点相对帧头的时间偏移是否满足您的算法，运动轨迹是由会漂移很大？

zfc-zfc, 2023-07-04:
> 这个当时场景前方5m左右有墙面。您那边分析我的点云信息是否正确，缺少ring这个参数有影响吗？我的数据中每个点相对帧头的时间偏移是否满足您的算法，运动轨迹是由会漂移很大？
场景太过于狭窄会导致雷达容易退化，导致标定失败。至于后面的问题，需要先解决场景问题才能得出结论，建议拿到开阔的场景进行测试。

closed 2023-07-12 by zfc-zfc