hku-mars/LiDAR_IMU_Init #70: 测试效果不太良好

SlamSmallFish, 2023-07-04:
作者，您好！向您请教一个问题，我在测试您提供的mid360的数据的时候，一运行到 refinement 时数据就开始跑偏了，而且得到的外参不如初始化时准确，这是为什么呢？

SlamSmallFish, 2023-07-04:
[image]
[image]

zfc-zfc, 2023-07-06:
> 作者，您好！向您请教一个问题，我在测试您提供的mid360的数据的时候，一运行到 refinement 时数据就开始跑偏了，而且得到的外参不如初始化时准确，这是为什么呢？

你的参数是怎么样的呢？可以截一张yaml文件的图吗

closed 2023-07-14 by zfc-zfc

CharlieV5, 2023-07-14:
与[SlamSmallFish]的问题一样，XYZ三轴激励时地图正常，到online refinement轨迹和地图就飘了
经过研究yaml参数，发现要把mean_acc_norm设置为1（一开始设置的是9.805），表示livox内部自带的IMU
如图，供参考！

[image]
[image]

> SlamSmallFish

zfc-zfc, 2023-07-15:
> 与[SlamSmallFish]的问题一样，XYZ三轴激励时地图正常，到online refinement轨迹和地图就飘了 经过研究yaml参数，发现要把mean_acc_norm设置为1（一开始设置的是9.805），表示livox内部自带的IMU 如图，供参考！
> 
> [image] [image]
> 
> > SlamSmallFish

是的，我在README里面写的很清楚，Livox内置IMU需要填 1