hku-mars/LiDAR_IMU_Init #71: 加速度标定存在问题

WadeFrank, 2023-07-14:
首先非常感谢大佬的工作。
在尝试进行HAP激光雷达和外置IMU的标定时，似乎平移上存在一些问题，我们标定的场景是室内场景。
这是我的角速度和线加速度结果的截图，麻烦您帮我看看存在什么问题，非常感谢！
[image]
[image]

zfc-zfc, 2023-07-15:
> 首先非常感谢大佬的工作。 在尝试进行HAP激光雷达和外置IMU的标定时，似乎平移上存在一些问题，我们标定的场景是室内场景。 这是我的角速度和线加速度结果的截图，麻烦您帮我看看存在什么问题，非常感谢！ [image] [image]

看右下方这张图片，XY方向的加速度激励可能太小了，导致约束不足

WadeFrank, 2023-07-17:
那可以请问一下，为什么左下方图片中激光雷达里程计和IMU的X Y轴的加速度差别这么大吗？（可以明显看出一开始IMU的加速度的变化范围要比激光雷达里程计大）

zfc-zfc, 2023-07-17:
> 那可以请问一下，为什么左下方图片中激光雷达里程计和IMU的X Y轴的加速度差别这么大吗？（可以明显看出一开始IMU的加速度的变化范围要比激光雷达里程计大）

加速度的变化范围不同是很正常的。刚体上任意两点AB的加速度与角速度关系是不同的，角速度满足 R\omega_A = \omega_B，所以AB两点的角速度norm相等；但加速度不是，AB两点的加速度关系可以参考论文公式(21)，由于有角速度、角加速度和平移外参的存在，会导致AB的加速度norm不同。

zfc-zfc, 2023-07-17:
> 100

你觉得标定出来的平移有问题，具体而言和真值差了多少呢？

NHMMing, 2023-07-17:
> 

每次标定出的结果不太一致，实际测量值lidar到imu的距离(x,y,z)=(0.169, 0.002, 0.048)m。在实验过程中结果如下：(x,y,z)=(0.092, 0.046, 0.001)m；(x,y,z)=(0.081, 0.045, 0.044)m；(x,y,z)=(0.134, 0.017, 0.089)m等

WadeFrank, 2023-07-18:
我又重新测试了一下，发现在开始online refinement的瞬间，rviz总是会卡顿一下，然后位姿马上会有漂移，请问这是什么情况呢？有解决的办法吗？

WadeFrank, 2023-07-18:
还有请问一下，能否在代码中给定外参的平移部分，只标定外参的旋转部分？

huangwefeng121, 2023-07-18:
> 我又重新测试了一下，发现在开始online refinement的瞬间，rviz总是会卡顿一下，然后位姿马上会有漂移，请问这是什么情况呢？有解决的办法

这个漂移的问题我也遇到了，你那边有什么思路解决吗，是不是imu本身的偏差导致的

af-doom, 2023-08-04:
> > 
> 
> 每次标定出的结果不太一致，实际测量值lidar到imu的距离(x,y,z)=(0.169, 0.002, 0.048)m。在实验过程中结果如下：(x,y,z)=(0.092, 0.046, 0.001)m；(x,y,z)=(0.081, 0.045, 0.044)m；(x,y,z)=(0.134, 0.017, 0.089)m等

您好，您解决这个问题了吗

zfc-zfc, 2023-08-09:
> 还有请问一下，能否在代码中给定外参的平移部分，只标定外参的旋转部分？

旋转的标定本就与平移无关，具体请参考论文公式。

zfc-zfc, 2023-08-09:
> 我又重新测试了一下，发现在开始online refinement的瞬间，rviz总是会卡顿一下，然后位姿马上会有漂移，请问这是什么情况呢？有解决的办法吗？
陀螺仪和加速度计的单位正确吗？应该为rad/s和m/s^2。类似问题也可能导致refinement漂移，参考#43

xbnz-1997, 2023-12-22:
@zfc-zfc 感谢大佬精彩的作品！我在用该算法运行自采和你所上传的数据集标定avia和内置IMU外参时，得到的平移参数与avia用户手册中的偏差较大，请问可能是什么原因导致的呢？avia_50ms_indoor.bag运行后获得平移为（0.018253, -0.009849, -0.001007)m，而 avia用户手册中给的是(-41.65, -23.26, 28.4)mm。

85256638, 2024-09-18:
> @zfc-zfc 感谢大佬精彩的作品！我在用该算法运行自采和你所上传的数据集标定avia和内置IMU外参时，得到的平移参数与avia用户手册中的偏差较大，请问可能是什么原因导致的呢？avia_50ms_indoor.bag运行后获得平移为（0.018253, -0.009849, -0.001007)m，而 avia用户手册中给的是(-41.65, -23.26, 28.4)mm。

我也遇到了同样的问题，我得出的平移为：Translation LiDAR to IMU (meter)   = -0.000341  0.021774 -0.053424。 而Horizon用户手册给的是：
[image]

BloodLemonS, 2025-04-18:
> 我也遇到了同样的问题，我得出的平移为：Translation LiDAR to IMU (meter) = -0.000341 0.021774 -0.053424。 而Horizon用户手册给的是： [image]
请问这个问题最终解决了吗？我得出的平移也是存在一定的误差的

BloodLemonS, 2025-04-18:
> 我也遇到了同样的问题，我得出的平移为：Translation LiDAR to IMU (meter) = -0.000341 0.021774 -0.053424。 而Horizon用户手册给的是： [image]

请问这个问题最终解决了吗？我标定出来的结果在水平位置上也是有一定的偏移的。这个有办法解决吗？

hopenessQAQ, 2025-06-06:
> > 
> 
> 每次标定出的结果不太一致，实际测量值lidar到imu的距离(x,y,z)=(0.169, 0.002, 0.048)m。在实验过程中结果如下：(x,y,z)=(0.092, 0.046, 0.001)m；(x,y,z)=(0.081, 0.045, 0.044)m；(x,y,z)=(0.134, 0.017, 0.089)m等

你好，平移误差问题最后解决了么？我也遇到类似问题，每次标定的平移都不一样，