hku-mars/LiDAR_IMU_Init #77: 标定进度条跑太快，结果不太准确

YYY-CV, 2023-08-07:
感谢大佬的工作！我有个问题希望大佬不吝赐教：
我的雷达是MID-360，在室内场景采集数据标定。
[image]
但是我发现标定的时候XYZ三轴的进度条跑得快的有点异常（一个轴的进度条正常给激励跑完之后，剩下两个轴几乎同时在1s左右全部跑完，这样的话来不及给充足的激励就完成了标定，然后开始在线精化）。按照demo中的演示，我大概每个轴摆动一个来回初始标定就结束了，有时候甚至第三个轴还没有开始摆动标定就完成了。肉眼看建图效果尚可，但是我将结果绘图之后发现ACC图中Lidar和IMU曲线并不是非常贴合：
[image]
外参收敛效果也不是很好（这已经是我挑选的比较好的一组结果了，有时候平移外参在标定结束后依然没有收敛到一条水平线，仍然是倾斜直线）：
[image]

把标定参数放到fast-lio2中去跑精度好像并未提升太多，定位时仍然时不时有轨迹跳变和回偏。
我除了将yaml文件中的imu topic改为雷达发布的名称，以及将mean_acc_norm设置为1外，其余参数均按照您的代码中未做变动：
[image]
希望大佬助我解惑，已经困扰我一周多了，再次感谢大佬精彩的工作！！

zfc-zfc, 2023-08-09:
首先你需要确保你的点云原始频率 * cut_frame_num 约等于 50；如果原始频率为10Hz，那么cut_frame_num可以设置为5。标定结束太快的可能原因之一就是 “点云原始频率 * cut_frame_num” 数值太大，产生标定数据的速度太快。
将data_accumulate_length更改为更大的值可以延长初始化的时间，比如1000~1500。

YYY-CV, 2023-08-09:
谢谢大佬！我将data_accumulate_length调大之后确实标定结果好一些了，下面是标定的一些结果图：
[image]
[image]
[image]
不过我发现旋转外参的收敛程度好像不是很好，平移外参倒是基本上收敛了，请问这个是什么原因呢？

另外为了验证这组标定结果的好坏我在室内采集了两个不同的数据bag分别用于建图和定位，定位的结果如下：
[image]
[image]
上面是用标定后的参数跑出来的结果，下面是把平移外参置为0，旋转外参置为单位阵，时间偏移置为0的运行结果，两组实验都把fast-lio2的在线优化外参关掉了。从图上的轨迹来看，肉眼很难看出区别，定位轨迹本身也没有明显的回偏和跳变，请问这是否说明标定结果没有那么准确，或者所实验的场景过于简单了呢？如果是场景问题，该采集何种数据才能区分开两者的精度差别呢？