hku-mars/LiDAR_IMU_Init #80: Degrees/sec to Rad/sec

ManChrys, 2023-09-07:
hello if i want to convert degrees to rad/s should i add these lines here? its the correct place? should i convert the velocities before pushing the message to the buffer right ? i saw issue #43, but i would like this clarification .

    //IMU Time Compensation
    msg->header.stamp = ros::Time().fromSec(msg->header.stamp.toSec() - timediff_imu_wrt_lidar - time_lag_IMU_wtr_lidar);
    double timestamp = msg->header.stamp.toSec();

    if (timestamp < last_timestamp_imu) {
        ROS_WARN("IMU loop back, clear IMU buffer.");
        imu_buffer.clear();
        Init_LI->IMU_buffer_clear();
    }

    last_timestamp_imu = timestamp;
    
    msg->angular_velocity.x /= (180/PI_M);
    msg->angular_velocity.y /= (180/PI_M);
    msg->angular_velocity.z /= (180/PI_M);

    imu_buffer.push_back(msg);
    
    Thanks in advance.

zfc-zfc, 2023-09-08:
Correct.

ManChrys, 2023-09-08:
Thank you my friend.

closed 2023-09-08 by ManChrys