hku-mars/LiDAR_IMU_Init #82: 加速度激励相关问题

Nothand0212, 2023-09-15:
您好，我看了很多Issues，都提到了“加速度激励”这个概念，而且都需要在标定完之后，通过绘图观察。
我想在代码里加入判断，一段时间内加速度激励不够大，就在终端里输出提示。
请问，对加速度激励这个概念，我该如何判断呢？相关的变量是什么？

万分感激，祝您成果多多

zfc-zfc, 2023-09-24:
> 您好，我看了很多Issues，都提到了“加速度激励”这个概念，而且都需要在标定完之后，通过绘图观察。 我想在代码里加入判断，一段时间内加速度激励不够大，就在终端里输出提示。 请问，对加速度激励这个概念，我该如何判断呢？相关的变量是什么？
> 
> 万分感激，祝您成果多多

加速度激励可以理解为，在各个方向上的加速度需要有较大的变化。一个比较简单的办法是，设置一个固定长度的buffer，存储过去一段时间的imu的加速度测量或者雷达里程计估计出的加速度。持续计算这个buffer内加速度的方差，大于一个阈值则认为加速度激励足够。

Cai-RS, 2024-05-07:
> > 您好，我看了很多Issues，都提到了“加速度激励”这个概念，而且都需要在标定完之后，通过绘图观察。 我想在代码里加入判断，一段时间内加速度激励不够大，就在终端里输出提示。 请问，对加速度激励这个概念，我该如何判断呢？相关的变量是什么？
> > 万分感激，祝您成果多多
> 
> 加速度激励可以理解为，在各个方向上的加速度需要有较大的变化。一个比较简单的办法是，设置一个固定长度的buffer，存储过去一段时间的imu的加速度测量或者雷达里程计估计出的加速度。持续计算这个buffer内加速度的方差，大于一个阈值则认为加速度激励足够。

你好！我想再请教一下，“持续计算这个buffer内加速度的方差”这一步，需要先将各个加速度测量值（是一个向量）统一到一个坐标系下吗？因为每个IMU的数据都是在前一个IMU时刻的坐标系下的，把它们直接相减似乎是不可取的？还是说加速度激励就必须得是这么直接计算和评估的？