hku-mars/LiDAR_IMU_Init #83: time lag lidar-imu

ManChrys, 2023-10-01:
hello i would like to ask whats the ideal time difference between lidar and imu for mapping (that means the best precision)? under second ? some miliseconds? for example i have 80 miliseconds. Is this diff big ?

Thanks in advance.

zfc-zfc, 2023-10-04:
> hello i would like to ask whats the ideal time difference between lidar and imu for mapping (that means the best precision)? under second ? some miliseconds? for example i have 80 miliseconds. Is this diff big ?
> 
> Thanks in advance.

Based on experience, the smaller the time difference, the higher the accuracy. 80 milliseconds is a relatively large time difference and can cause significant issues during high-speed motion. Typically, a time difference of within 5 milliseconds can generally meet most requirements.

ManChrys, 2023-10-04:
Thanks my friend !!

closed 2023-10-04 by ManChrys