hku-mars/LiDAR_IMU_Init #84: How use mid360 (livox_ros_driver2) if this software use livox_ros_driver?

FPSychotic, 2023-10-27:
Hi , I seen you given support to mid-360 using livox_ros_driver, but livox mid-360 , only can run with livox_ros_driver2. I know in the experimental phase it was using livox_ros_driver and livox SDK , but from a while I think , not sure if I'm wrong it only works quit livox_ros_driver2 and SDK2, so , how you use actually mid-360 with livox_ros_driver1? 
Also I think there was a change in the gravity processing and time, because them the release of livox_ros_driver2 how you deal with that ?
In my case, maybe I do something wrong, but I cannot use mid-360 with livox_ros_driver(1) as here is proposed or even Point-lio, how you do?

thanks in advance

zfc-zfc, 2023-10-29:
This software also supports livox_ros_driver2.

zfc-zfc, 2023-10-29:
And Livox-SDK2.

FPSychotic, 2023-10-29:
It said cannot fins livos_ros_driver, installed livox_ros_driver, but livox ros driver needs SDK1,installed SDK1, but my mid360 dont show points.
I cannot see any reference to livox_ros_driver2 in the code, so for that my question,how you do make it work, because I cannot .
Thanks in advance

zfc-zfc, 2023-10-31:
> It said cannot find livos_ros_driver, installed livox_ros_driver, but livox ros driver needs SDK1,installed SDK1, but my mid360 dont show points. I cannot see any reference to livox_ros_driver2 in the code, so for that my question,how you do make it work, because I cannot. Thanks in advance

You don't need to change anything of this repo. For mid360 LiDAR, you should:
1. Install livox-SDK2
2. Open a new workspace "livox2_ws" and place livox_ros_driver2 in it.
3. Compile livox2_ws and modify LiDAR IP following https://github.com/Livox-SDK/livox_ros_driver2.
4. Run "roslaunch livox_ros_driver2 msg_MID360.launch" and make sure points are generated.
5. Run lidar_imu_init.

FPSychotic, 2023-10-31:
Thanks by your attention

closed 2023-10-31 by FPSychotic