hku-mars/LiDAR_IMU_Init #90: How to use this program to calibrate mid360 parameters?

wyb17js, 2023-12-27:
As shown in the title, we use ros_driver2 to drive mid360, release point cloud in customized format, and try to use this program to match the timestamp of laser and imu for mid360. We tilted the radar according to the method in the video to complete the progress in the three directions of xyz respectively, and got the final result. But each time we get a different result, how can I evaluate the final result?

zfc-zfc, 2023-12-31:
Getting different results each time is normal, as any algorithm calibration comes with errors. If the differences in your results are not significant (such as a rotation of 1 degree or a translation of approximately 2 cm), you can use the average value by calibrating several times. However, if the differences are substantial, it is recommended to calibrate again..

wyb17js, 2024-01-02:
> Getting different results each time is normal, as any algorithm calibration comes with errors. If the differences in your results are not significant (such as a rotation of 1 degree or a translation [...]

What about the time stamp gap? Is the timestamp from each correction also averaged? I saw that the mid360 official website has its imu radar relative position information, so my key question is to get the time difference between them.

zfc-zfc, 2024-01-04:
> What about the time stamp gap? Is the timestamp from each correction also averaged? I saw that the mid360 official website has its imu radar relative position information, so my key question is to [...]

“Is the timestamp from each correction also averaged？” -- YES.
Under normal circumstances, the time lag between multiple calibrations of the mid360 and IMU should not vary significantly, usually smaller than 5ms if the parameters are tuned well.