hku-mars/LiDAR_IMU_Init #93: mid360 yaml配置问题

batmelody, 2024-01-25:
common:
    lid_topic:  "/livox/lidar"
    imu_topic:  "/mavros/imu/data_raw"

preprocess:
    lidar_type: 1                # Livox series LiDAR
    feature_extract_en: false
    scan_line: 6
    blind: 1

initialization:
    cut_frame_num: 5            # must be positive integer
    orig_odom_freq: 10           # original Lidar frequency
    mean_acc_norm: 9.805         # 1 for livox built-in IMU
    online_refine_time: 20     # seconds
    data_accum_length: 500
    Rot_LI_cov: [ 0.00005, 0.00005, 0.00005 ]
    Trans_LI_cov: [ 0.00001, 0.00001, 0.00001 ]

mapping:
    filter_size_surf: 0.05
    filter_size_map: 0.15
    gyr_cov: 50
    acc_cov: 2
    b_acc_cov: 0.0001
    b_gyr_cov: 0.0001
    det_range: 100.0

publish:
    path_en:  true
    scan_publish_en:  true       # false: close all the point cloud output
    dense_publish_en: true       # false: low down the points number in a global-frame point clouds scan.
    scan_bodyframe_pub_en: false  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: false
    interval: -1                 # how many LiDAR frames saved in each pcd file;
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.

zfc-zfc, 2024-01-26:
> common: lid_topic: "/livox/lidar" imu_topic: "/mavros/imu/data_raw"
> 
> preprocess: lidar_type: 1 # Livox series LiDAR feature_extract_en: false scan_line: 6 blind: 1
> 
> initialization: cut_frame_num: 5 # must be positive integer orig_odom_freq: 10 # original Lidar frequency mean_acc_norm: 9.805 # 1 for livox built-in IMU online_refine_time: 20 # seconds data_accum_length: 500 Rot_LI_cov: [ 0.00005, 0.00005, 0.00005 ] Trans_LI_cov: [ 0.00001, 0.00001, 0.00001 ]
> 
> mapping: filter_size_surf: 0.05 filter_size_map: 0.15 gyr_cov: 50 acc_cov: 2 b_acc_cov: 0.0001 b_gyr_cov: 0.0001 det_range: 100.0
> 
> publish: path_en: true scan_publish_en: true # false: close all the point cloud output dense_publish_en: true # false: low down the points number in a global-frame point clouds scan. scan_bodyframe_pub_en: false # true: output the point cloud scans in IMU-body-frame
> 
> pcd_save: pcd_save_en: false interval: -1 # how many LiDAR frames saved in each pcd file; # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.
> 
> 您好，我使用的激光雷达是mid360，这是我截取您yaml文件里的配置。这里的acc_cov与gyr_cov是否过大了。还有其它值是否有什么需要修改的？

就是要很大，否则EKF本身的滤波器延迟会影响时差标定结果。其他 注意一下地图分辨率就好

batmelody, 2024-02-02:
谢谢回复，当我使用一个很大的cov之后，我的标定建图会发生错层，抖动；把cov减小将恢复正常，请问这种情况下要如何去调整这个cov呢？