hku-mars/LiDAR_IMU_Init #95: Lidar frame splitting and filter size setting questions

laomeng0703, 2024-02-01:
Hello,
Thank you for your great work. I have some questions about lidar frame splitting and filter size. In readme.md, your recommendation for lidar frame cut_frame_num is to set velodyne to 3, but when I actually use velodyne, if cut_frame_num is set to 3, the map will become a mess at the beginning, but when I set it If cut_frame_num is set to 10, the result is good. Is this related to the frequency of imu? My imu frequency is set to 100hz. Would it be better to set the lidar frame number and the imu frequency to a certain ratio?
And, how to define indoor and outdoor? If I'm in a large room such as a factory, does that mean I need to set the filter size to outdoor?
Thanks again for your work, what a perfect map!
[image]

SangHunHan92, 2024-02-02:
Hi! How did you get full 3D map?
Just run ```roslaunch lidar_imu_init xxx.launch``` ??

zfc-zfc, 2024-02-03:
> what
It seems strange. If a point cloud frame is splitted into 10 parts based on timestamps, each frame would only have a FoV of 36°, which could easily lead to degradation of the LiDAR. Can you share your bag with me so that I can investigate this issue?

laomeng0703, 2024-02-05:
> > what
> > It seems strange. If a point cloud frame is splitted into 10 parts based on timestamps, each frame would only have a FoV of 36°, which could easily lead to degradation of the LiDAR. Can you share your bag with me so that I can investigate this issue?

Sure. Here is the link: https://ejnu-my.sharepoint.com/:u:/g/personal/208056_jnu_ac_kr/EdDmuEduht1AjGrwudlbcxIBwCTf1ZLv9SfzBNsx5070QA?e=tFlagv

Thank you.

HamzaCetin0, 2024-07-23:
Hi @laomeng0703 .can you tell me the other parameter that you specified on the .yaml file