hku-mars/LiDAR_IMU_Init #96: IMU Selection

aditdoshi333, 2024-02-07:
Hello,

Thank you for sharing your appreciation for the work. I've followed many of your videos to compare SLAM algorithms and need your help in selecting a SLAM algorithm for my use case.

I have Livox Avia and Livox Mid 360, and we have experimented with the following SLAM algorithms. We need a SLAM algorithm for mapping small and large rooms. However, we are facing issues when generating point clouds for complex spaces that combine large areas, narrow paths, and small rooms.

I am considering trying two different approaches: one with a dual LiDAR setup (Avia + Mid and Mid + Mid) and the other with a Mid360 + external IMU. However, I'm not sure which IMU to try.

Could you please share your feedback on which IMU I should consider?

Thank you very much.

laomeng0703, 2024-02-22:
3DM-GX5-25 is good in my experience.

zfc-zfc, 2024-02-25:
> Hello,
>
> Thank you for sharing your appreciation for the work. I've followed many of your videos to compare SLAM algorithms and need your help in selecting a SLAM algorithm for my use case.
>
> I have [...]

Why don't you use the internal IMU of Mid360?

satyajit-ink, 2024-03-05:
I think you should not rely on IMU, internal should be good enough.