hku-mars/LiDAR_IMU_Init #102: Using Docker. VLP16 and PX4 IMU. Initialization starts to fill when LiDAR+IMU are static and leads to Strange behaviour.

Moraisgit, 2024-04-23:
Hi! Thank you so much for the great work. I hope you can help me.

I'm using a VLP16 LiDAR and a PX4 IMU and I've setted up the package with Docker. Below I attach a video showing what happens in my setup.

On the left terminal I just have the connection from my PC to a Jackal robot which has the VLP16 and PX4 mounted on it (the process running is just a `roslaunch` to enable the IMU). On the right terminal I have the Docker container running in which I exported the `roscore` of the Jackal robot and for that reason I `echo` the LiDAR and IMU topics that I need.

Then I run the actual package with the `velodyne.launch`file and you can see what happens in the video. I'm using real time data, not a bag file. Keep in mind I did not move the entire robot, keeping the LiDAR and IMU static. The initialization just starts to fill and on rviz it displays strange behaviour:

https://github.com/hku-mars/LiDAR_IMU_Init/assets/114250545/450c75fc-ed66-48da-bdd4-e276232b61c5

I have read Issues #9 (I changed the parameters that you advised to change, below I attach a picture my `velodyne.yaml` file); #43 (I verified and is shown in the video that the angular velocity of input seems to be in rad/s).
[image]

If you could help me it would be much appreciated. Thank you very much in advance!

zfc-zfc, 2024-04-27:
> @zfc-zfc Hi! Thank you so much for the great work. I hope you can help me.
> 
> I'm using a VLP16 LiDAR and a PX4 IMU and I've setted up the package with Docker. Below I attach a video showing what happens in my setup.
> 
> On the left terminal I just have the connection from my PC to a Jackal robot which has the VLP16 and PX4 mounted on it (the process running is just a `roslaunch` to enable the IMU). On the right terminal I have the Docker container running in which I exported the `roscore` of the Jackal robot and for that reason I `echo` the LiDAR and IMU topics that I need.
> 
> Then I run the actual package with the `velodyne.launch`file and you can see what happens in the video. I'm using real time data, not a bag file. Keep in mind I did not move the entire robot, keeping the LiDAR and IMU static. The initialization just starts to fill and on rviz it displays strange behaviour:
> 
>  423d621f-29eb-4a29-b7b5-5bba753e51b2.mp4 
> I have read Issues #9 (I changed the parameters that you advised to change, below I attach a picture my `velodyne.yaml` file); #43 (I verified and is shown in the video that the angular velocity of input seems to be in rad/s). ![image](https://private-user-images.githubusercontent.com/114250545/324978343-7fa74b81-b86c-4706-a683-d4f70a47d8f9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQyMDcyMTIsIm5iZiI6MTcxNDIwNjkxMiwicGF0aCI6Ii8xMTQyNTA1NDUvMzI0OTc4MzQzLTdmYTc0YjgxLWI4NmMtNDcwNi1hNjgzLWQ0ZjcwYTQ3ZDhmOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDI3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyN1QwODM1MTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kMWI4MGY3MDYxYjhkNjhmNWIxM2E1NzY4NTMwNzgxYzMyY2JiNDk2OTg3NWVmY2U3ZGM2M2NmMjYxOGQ0NDAyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.YD8Uzs_8Be7jCcoftOXjIpz8n7ca290SI2wRjhYCToE)
> 
> If you could help me it would be much appreciated. Thank you very much in advance!

My suggestion: 1. cut_frame_num = 2; filter_size_surf = 0.2; filter_size_map = 0.3
If the above modification still doesn't work, please try a larger scene.

HamzaCetin0, 2024-07-25:
> @zfc-zfc Hi! Thank you so much for the great work. I hope you can help me.
>
> I'm using a VLP16 LiDAR and a PX4 IMU and I've setted up the package with Docker. Below I attach a video showing what [...]

Hi @Moraisgit. I have the same problem. Did you solve it initialization problem.

Moraisgit, 2024-07-26:
> Hi @Moraisgit. I have the same problem. Did you solve it initialization problem.

Hi @HamzaCetin0! I am sorry but I did not solve the problem. I ended up switching to a OS1 LiDAR and using [FAST-LIO](https://github.com/hku-mars/FAST_LIO) with the LiDAR's internal IMU.