hku-mars/LiDAR_IMU_Init #103: 外参标定重复性问题

hwh666, 2024-05-14:
你好，当 cut_frame_num，filter_size_surf_min，filter_size_map_min等参数改变时，每次外参标定的结果都会有偏差，rpy角会偏1-2°，时延会偏10ms以内，想请教下：
1. 这些参数我如何判断给到了最优
2. 不同的配置文件标定出的结果如何判断那个结果最优

zfc-zfc, 2024-05-14:
> 你好，当 cut_frame_num，filter_size_surf_min，filter_size_map_min等参数改变时，每次外参标定的结果都会有偏差，rpy角会偏1-2°，时延会偏10ms以内，想请教下：
> 
> 1. 这些参数我如何判断给到了最优
> 2. 不同的配置文件标定出的结果如何判断那个结果最优

你好，重复性问题确实是存在的，这主要是由于不同的参数会影响LiDAR Odometry的结果，从而导致标定的结果差异。当然许多算法都会有此现象，即参数影响结果。理论上说，在激励充分、场景特征丰富的情况下，改变这几个参数对标定结果的影响会很小，标定的结果能满足绝大部分使用需求。
对于问题1，我认为当然是看标定结果：标定结果越准确，参数就越好，这也就回到了问题2.
对于问题2，在有真值的情况下和真值对比可以看出标定误差，从而衡量结果的优劣；但往往大部分情况下没有真值存在，我建议可以在室内场景标定，在refine完成后，即FAST-LIO2阶段，给予设备猛烈的激励（猛甩设备），看看不同结果得到的建图的一致性（墙面厚度）。

hwh666, 2024-05-14:
> 你好，重复性问题确实是存在的，这主要是由于不同的参数会影响LiDAR Odometry的结果，从而导致标定的结果差异。当然许多算法都会有此现象，即参数影响结果。理论上说，在激励充分、场景特征丰富的情况下，改变这几个参数对标定结果的影响会很小，标定的结果能满足绝大部分使用需求。 对于问题1，我认为当然是看标定结果：标定结果越准确，参数就越好，这也就回到了问题2. [...]

好的，感谢您的回复，我再试一试

closed 2024-05-14 by hwh666