hku-mars/LiDAR_IMU_Init #113: Velodyne lidar misscalibation..

HamzaCetin0, 2024-09-03:
Hi, thank you for your amazing work about lidar, imu calibration. I have a Velodyne VLP16 (10Hz)  Lidar and Xsens Mti-630 IMU(200Hz). After the initialization I can get the results.txt but I think there are some problems in this file. The values are not correct. When I run the matlab code, angular velocity and linear acceleration curves are not aligned. If I send you the .bag file could you check for me? I also added the configuration file down below. I tried your suggestions( e.g. cut_frame_num = 3) about velodynes on the configuration file but when I changed this value to 3 the map went away suddenly. I think the ideal parameter is 10 for my setup. Could you please help me? 

Best regards.

Hamza.

Bag File Drive Link: [https://drive.google.com/file/d/172WOOn8HwY_lxrHyAbD8LaLRnC8q9jUK/view](url)
[image]
[image]

85256638, 2024-09-20:
#114 similar to the issue that I encountered, the liner accelerations looks not correct, hope the author could shed some light.

HamzaCetin0, 2025-02-01:
@85256638 Hi. Did you solve the problem? If so, how did you solve the problem?