hku-mars/LiDAR_IMU_Init #114: Livox Horizon反复多次标定，内置IMU的线加速度比激光雷达的线加速度差别非常大，几乎无法重合

85256638, 2024-09-19:
感谢大佬发布高质量的作品。我是用Horizon和其内置的imu进行标定，室外场景，每次激光雷达的线加速度比IMU的线加速度大非常多：
[image]
[image]

校准结果如下：
`Initialization result:
Rotation LiDAR to IMU (degree)     = -1.148407  0.976242  0.077176
Translation LiDAR to IMU (meter)   = -0.079951  0.089463 -0.023574
Time Lag IMU to LiDAR (second)     = -0.008057
Bias of Gyroscope  (rad/s)         = -0.014254  0.000652  0.017108
Bias of Accelerometer (meters/s^2) = -0.010185  0.009784 -0.010027
Gravity in World Frame(meters/s^2) = -0.548225  0.268300 -9.790994

Homogeneous Transformation Matrix from LiDAR to IMU: 
 0.999854 -0.001688  0.017006 -0.079951
 0.001347  0.999798  0.020064  0.089463
-0.017037 -0.020038  0.999654 -0.023574
 0.000000  0.000000  0.000000  1.000000

Refinement result:
Rotation LiDAR to IMU (degree)     = -2.308273  2.146560  0.307814
Translation LiDAR to IMU (meter)   =  0.014408 -0.026519 -0.045003
Time Lag IMU to LiDAR (second)     = -0.008057
Bias of Gyroscope  (rad/s)         =  0.002983 -0.000369 -0.002918
Bias of Accelerometer (meters/s^2) = -0.000807  0.009512 -0.047373
Gravity in World Frame(meters/s^2) = -0.536506  0.255392 -9.765394

Homogeneous Transformation Matrix from LiDAR to IMU: 
 0.999284 -0.006876  0.037206  0.014408
 0.005368  0.999166  0.040474 -0.026519
-0.037453 -0.040245  0.998488 -0.045003
 0.000000  0.000000  0.000000  1.000000
`

另外一组同样的设备，同样的场景校准结果如下：
[image]
[image]

校准结果如下：

`Initialization result:
Rotation LiDAR to IMU (degree)     = -0.648640  0.525992 -0.899704
Translation LiDAR to IMU (meter)   = -0.125558  0.074944 -0.008613
Time Lag IMU to LiDAR (second)     = -0.008028
Bias of Gyroscope  (rad/s)         = 0.003915 0.001149 0.004719
Bias of Accelerometer (meters/s^2) = -0.010200 -0.009895 -0.004655
Gravity in World Frame(meters/s^2) = -0.074485  0.187377 -9.807927

Homogeneous Transformation Matrix from LiDAR to IMU: 
 0.999835  0.015596  0.009355 -0.125558
-0.015700  0.999814  0.011174  0.074944
-0.009179 -0.011319  0.999894 -0.008613
 0.000000  0.000000  0.000000  1.000000

Refinement result:
Rotation LiDAR to IMU (degree)     = -1.405847  1.788843  0.014381
Translation LiDAR to IMU (meter)   = -0.032261  0.003270 -0.116524
Time Lag IMU to LiDAR (second)     = -0.008028
Bias of Gyroscope  (rad/s)         = 0.000219 0.000296 0.001392
Bias of Accelerometer (meters/s^2) = -0.013455 -0.018133 -0.038624
Gravity in World Frame(meters/s^2) = -0.037713  0.161751 -9.782781

Homogeneous Transformation Matrix from LiDAR to IMU: 
 0.999513 -0.001017  0.031198 -0.032261
 0.000251  0.999699  0.024540  0.003270
-0.031214 -0.024520  0.999212 -0.116524
 0.000000  0.000000  0.000000  1.000000
`

我的horizon.yaml 配置文件如下：

`common:
    lid_topic:  "/livox/lidar"
    imu_topic:  "/livox/imu"

preprocess:
    lidar_type: 1                # Livox series LiDAR
    feature_extract_en: false
    scan_line: 6
    blind: 1

initialization:
    cut_frame_num: 5 # must be positive integer
    orig_odom_freq: 10
    mean_acc_norm: 1 # 1: for livox built-in IMU
    online_refine_time: 20.0
    data_accum_length: 500
    Rot_LI_cov: [ 0.00005, 0.00005, 0.00005 ]
    Trans_LI_cov: [ 0.00001, 0.00001, 0.00001 ]

mapping:
    filter_size_surf: 0.3
    filter_size_map: 0.4
    gyr_cov: 50
    acc_cov: 2
    b_acc_cov: 0.0001
    b_gyr_cov: 0.0001
    det_range:     260.0

publish:
    path_en:  true
    scan_publish_en:  true       # false: close all the point cloud output
    dense_publish_en: true       # false: low down the points number in a global-frame point clouds scan.
    scan_bodyframe_pub_en: false  # true: output the point cloud scans in IMU-body-frame

pcd_save:
    pcd_save_en: true
    interval: -1                 # how many LiDAR frames saved in each pcd file; 
                                 # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.`

希望大佬指点迷津，不吝赐教！

85256638, 2024-09-20:
汇报一下：我又重新看了下大神的视频，我发现我的移动方式有问题，不是视频中的这个样子，因此我又重新按照视频里的移动轨迹，重新做了5次校准。结果好了不少，但是还是不够理想：

[image]

[image]

[image]

[image]

[image]

[image]

zfc-zfc, 2024-11-25:
> 汇报一下：我又重新看了下大神的视频，我发现我的移动方式有问题，不是视频中的这个样子，因此我又重新按照视频里的移动轨迹，重新做了5次校准。结果好了不少，但是还是不够理想： [...]

加速度的质量很依赖于LO的精度，而LO的精度比较依赖场景，可以尽量选一个平面特征多的场景，比如地下车库，建筑物内部。可以看看你的yaml参数表吗？

sanbens1234, 2024-11-29:
你好,请教一下关于horizon用lidar-imu-init的方法标定,我启动roslaunch lidar_imu_init livox_horizon.launch后,为什么很快三个方向的进度条就都满了?雷达还在原地,根本没有给他三个方向的激励.输出的标定结果也显然完全不对.感谢!

85256638, 2024-12-04:
> 你好,请教一下关于horizon用lidar-imu-init的方法标定,我启动roslaunch lidar_imu_init livox_horizon.launch后,为什么很快三个方向的进度条就都满了?雷达还在原地,根本没有给他三个方向的激励.输出的标定结果也显然完全不对.感谢!

有可能是你的场景太小了，近处有障碍物，试试放在室外？

85256638, 2024-12-04:
> 加速度的质量很依赖于LO的精度，而LO的精度比较依赖场景，可以尽量选一个平面特征多的场景，比如地下车库，建筑物内部。可以看看你的yaml参数表吗？

哎…我系统重装了…我改天再装一次测试一下。

sanbens1234, 2024-12-04:
> 哎…我系统重装了…我改天再装一次测试一下。

请问你最后跑出来point-lio的结果怎么样?我看x和y基本没问题,z轴漂移的问题比较严重.另外,标定imu与lidar的时间差似乎没有意义,因为我看源代码,那个参数time_diff_lidar_to_imu最后没有被赋值并使用

85256638, 2024-12-13:
> 请问你最后跑出来point-lio的结果怎么样?我看x和y基本没问题,z轴漂移的问题比较严重.另外,标定imu与lidar的时间差似乎没有意义,因为我看源代码,那个参数time_diff_lidar_to_imu最后没有被赋值并使用

我没有测试过point-lio，有机会测试

85256638, 2024-12-13:
> 加速度的质量很依赖于LO的精度，而LO的精度比较依赖场景，可以尽量选一个平面特征多的场景，比如地下车库，建筑物内部。可以看看你的yaml参数表吗？

谢谢大佬，我这回换成地下停车场了，线加速度貌似有改善，但是还是不如别人测试出的结果：

[image]

测试1：

[image]

[image]

Initialization Result 1：

```
Initialization result:
Rotation LiDAR to IMU (degree)     = -1.651767 -0.310123  0.687265
Translation LiDAR to IMU (meter)   = -0.110982 -0.013314 -0.034078
Time Lag IMU to LiDAR (second)     = -0.002122
Bias of Gyroscope  (rad/s)         = 0.001023 0.003493 0.000672
Bias of Accelerometer (meters/s^2) =  0.009924  0.009901 -0.007683
Gravity in World Frame(meters/s^2) = -0.626068  0.481949 -9.778132

Homogeneous Transformation Matrix from LiDAR to IMU: 
 0.999913 -0.011833 -0.005755 -0.110982
 0.011994  0.999515  0.028756 -0.013314
 0.005412 -0.028822  0.999570 -0.034078
 0.000000  0.000000  0.000000  1.000000

Refinement result:
Rotation LiDAR to IMU (degree)     = -1.509254 -0.334247  0.964456
Translation LiDAR to IMU (meter)   =  0.036230  0.024046 -0.013446
Time Lag IMU to LiDAR (second)     = -0.002122
Bias of Gyroscope  (rad/s)         =  0.000203 -0.001716  0.000412
Bias of Accelerometer (meters/s^2) =  0.019986  0.019764 -0.030311
Gravity in World Frame(meters/s^2) = -0.617381  0.476924 -9.759885

Homogeneous Transformation Matrix from LiDAR to IMU: 
 0.999841 -0.016671 -0.006274  0.036230
 0.016831  0.999514  0.026235  0.024046
 0.005833 -0.026336  0.999636 -0.013446
 0.000000  0.000000  0.000000  1.000000
```

测试2：

[image]

[image]

[image]

Initialization Result 2：

```
Rotation LiDAR to IMU (degree)     = -0.826641 -0.225618 -0.182743
Translation LiDAR to IMU (meter)   =  0.005643  0.086310 -0.078526
Time Lag IMU to LiDAR (second)     = -0.001803
Bias of Gyroscope  (rad/s)         = -0.003765 -0.003836 -0.002947
Bias of Accelerometer (meters/s^2) =  0.010071  0.009823 -0.010104
Gravity in World Frame(meters/s^2) = -0.524762  0.383424 -9.788448

Homogeneous Transformation Matrix from LiDAR to IMU: 
 0.999987  0.003246 -0.003891  0.005643
-0.003189  0.999891  0.014439  0.086310
 0.003937 -0.014426  0.999888 -0.078526
 0.000000  0.000000  0.000000  1.000000

Refinement result:
Rotation LiDAR to IMU (degree)     = -0.308096 -0.541690  0.925623
Translation LiDAR to IMU (meter)   =  0.076496  0.050379 -0.016162
Time Lag IMU to LiDAR (second)     = -0.001803
Bias of Gyroscope  (rad/s)         =  0.002440 -0.002316 -0.000287
Bias of Accelerometer (meters/s^2) = -0.003373  0.011977 -0.019799
Gravity in World Frame(meters/s^2) = -0.476321  0.350985 -9.767144

Homogeneous Transformation Matrix from LiDAR to IMU: 
 0.999825 -0.016102 -0.009539  0.076496
 0.016153  0.999856  0.005223  0.050379
 0.009453 -0.005377  0.999941 -0.016162
 0.000000  0.000000  0.000000  1.000000

```

shixiaosongye, 2024-12-14:
您好请问下这类数据对比图您是怎么绘制出来的呀？
[image]

85256638, 2024-12-15:
@shixiaosongye 项目目录里面有个`matlab_code` 和 `log `文件夹。 用matlab_code里面的代码来读取log里面的标定结果就可以绘出上面的图。 如果运行code有错误的话，可以让chatgpt小改一下就行了。

85256638, 2024-12-15:
> > 你好,请教一下关于horizon用lidar-imu-init的方法标定,我启动roslaunch lidar_imu_init livox_horizon.launch后,为什么很快三个方向的进度条就都满了?雷达还在原地,根本没有给他三个方向的激励.输出的标定结果也显然完全不对.感谢!
> 
> 有可能是你的场景太小了，近处有障碍物，试试放在室外？

你用的是激光雷达内置的IMU还是外置的IMU？

zfc-zfc, 2024-12-16:
> > > 汇报一下：我又重新看了下大神的视频，我发现我的移动方式有问题，不是视频中的这个样子，因此我又重新按照视频里的移动轨迹，重新做了5次校准。结果好了不少，但是还是不够理想：
> > > ![image](https://private-user-images.githubusercontent.com/3270742/369258087-ce615fda-6cbb-4f41-a445-9059acb71ac7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI1Mzc1OTksIm5iZiI6MTczMjUzNzI5OSwicGF0aCI6Ii8zMjcwNzQyLzM2OTI1ODA4Ny1jZTYxNWZkYS02Y2JiLTRmNDEtYTQ0NS05MDU5YWNiNzFhYzcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMjVUMTIyMTM5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9M2FmNmU4NTI4Yzk3OGJiZjkxYTlkOTk4YjllYTA5ODAzOGRlYjJjMzgxYmQ0OGI2ZjE0OGI2NDMzMzQwMDcxOSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.9kREWd3ihFaQvVVnq8zo9kWfb62XaYQpWEBeJnbeuko)
> > > ![image](https://private-user-images.githubusercontent.com/3270742/369258151-9699b2a2-99c7-4105-8049-7bb2ada75443.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI1Mzc1OTksIm5iZiI6MTczMjUzNzI5OSwicGF0aCI6Ii8zMjcwNzQyLzM2OTI1ODE1MS05Njk5YjJhMi05OWM3LTQxMDUtODA0OS03YmIyYWRhNzU0NDMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMjVUMTIyMTM5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTY2YzljODk0NGY1ZDEyNGFjNmFlMTU5MTFhOGQ2Njk5ZjNjNWFiOTViNDZkM2NmNWJiNjRkMTI0OGE2ZDM3NiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.MYoMCd7mwpw5EHBZUGHCGP6CqbQCNv32C1V3ag2Q_6Y)
> > > ![image](https://private-user-images.githubusercontent.com/3270742/369259110-960155a6-9d9e-4272-af0f-7cfd07d4d1e9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI1Mzc1OTksIm5iZiI6MTczMjUzNzI5OSwicGF0aCI6Ii8zMjcwNzQyLzM2OTI1OTExMC05NjAxNTVhNi05ZDllLTQyNzItYWYwZi03Y2ZkMDdkNGQxZTkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMjVUMTIyMTM5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzI1ZGJhYjQ4ZTFlYjk0N2NjYTk3NGVlYzQzMWYyOWJhNzk1ZDM0NzQ0YWI1ODI2MjU3YjQzNWNiN2UxNWY5ZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.GQR2OXT2jiweNjzJAYVwEgegKUUh__xs7B4Kd1MQjWw)
> > > ![image](https://private-user-images.githubusercontent.com/3270742/369259388-e3889c96-bd54-40c0-91fd-2f360fcd407a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI1Mzc1OTksIm5iZiI6MTczMjUzNzI5OSwicGF0aCI6Ii8zMjcwNzQyLzM2OTI1OTM4OC1lMzg4OWM5Ni1iZDU0LTQwYzAtOTFmZC0yZjM2MGZjZDQwN2EucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMjVUMTIyMTM5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjM2MDNjZGQzMDZjY2M2ZDMxYjg4OTQzYWFmYjU2MjY2MDI5NDJhYTAzY2FlYjZjMDIyZTQzNGQ4ZjJkNmQwMCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.81h0eMNuZaJXytPcfBQU8N08exe07kjhVJGbve-AKX8)
> > > ![image](https://private-user-images.githubusercontent.com/3270742/369259741-d7d92bf3-0048-47a9-89e5-5c3aefe96ab1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI1Mzc1OTksIm5iZiI6MTczMjUzNzI5OSwicGF0aCI6Ii8zMjcwNzQyLzM2OTI1OTc0MS1kN2Q5MmJmMy0wMDQ4LTQ3YTktODllNS01YzNhZWZlOTZhYjEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMjVUMTIyMTM5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTJmOGMyOTNlNTQ3ZDk1YWExZGNjMTAwNGE3YTRjZTE2NWMxZDFlZmVkMTlhOTgzYjViMTA4NmY5ODIzOWIzNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.6TZq7Xn3TKKPkLXRi-ojO_qEZlNmf1jVB8cKrOjpGiE)
> > > ![image](https://private-user-images.githubusercontent.com/3270742/369259974-81239e96-09ec-4df6-9ee4-d865764bf5c8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzI1Mzc1OTksIm5iZiI6MTczMjUzNzI5OSwicGF0aCI6Ii8zMjcwNzQyLzM2OTI1OTk3NC04MTIzOWU5Ni0wOWVjLTRkZjYtOWVlNC1kODY1NzY0YmY1YzgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTEyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDExMjVUMTIyMTM5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MjQ3ODEwMGZiYzk1ZjVmMzUzZjIwMDYzMTIyOWQ5NTEwYTFkOGE1Mzc5ODliZmVjYTgzMDBjY2QyNDYyYzIwMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.MV42y-iBr3d1RZHFEbWCq15K0WzgqImbZZ8bHa20aMg)
> > 
> > 
> > 加速度的质量很依赖于LO的精度，而LO的精度比较依赖场景，可以尽量选一个平面特征多的场景，比如地下车库，建筑物内部。可以看看你的yaml参数表吗？
> 
> 谢谢大佬，我这回换成地下停车场了，线加速度貌似有改善，我总共测试了两次。但是还是不如别人测试出的结果：
> 
> **测试1：**
> 
> ![image](https://private-user-images.githubusercontent.com/3270742/395438925-17e25156-f8da-4567-9818-91707a21a3a1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQzMjkxODgsIm5iZiI6MTczNDMyODg4OCwicGF0aCI6Ii8zMjcwNzQyLzM5NTQzODkyNS0xN2UyNTE1Ni1mOGRhLTQ1NjctOTgxOC05MTcwN2EyMWEzYTEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0 [... cut ...]
>     dense_publish_en: true       # false: low down the points number in a global-frame point clouds scan.
>     scan_bodyframe_pub_en: false  # true: output the point cloud scans in IMU-body-frame
> 
> pcd_save:
>     pcd_save_en: true
>     interval: -1                 # how many LiDAR frames saved in each pcd file; 
>                                  # -1 : all frames will be saved in ONE pcd file, may lead to memory crash when having too much frames.
> ```
> 
> @zfc-zfc

可以看看initialization阶段的点云地图是否清晰，如果不清晰，可能需要调整下LO的参数。
另外，把data_accum_length调大到1500试试