hku-mars/LiDAR_IMU_Init #115: States.h 头文件无法找到

MagicShawn, 2024-11-19:
LiDAR_IMU_Init/src/IMU_Processing.hpp:27:10: fatal error: lidar_imu_init/States.h: No such file or directory
 #include <lidar_imu_init/States.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~

blue-stone-j, 2024-11-20:
This header file id generated from a .msg file, which locates at packge_folder/msg/States.h . In common cases, this header will be generated during compilation.

zfc-zfc, 2024-11-25:
> LiDAR_IMU_Init/src/IMU_Processing.hpp:27:10: fatal error: lidar_imu_init/States.h: No such file or directory #include <lidar_imu_init/States.h> ^~~~~~~~~~~~~~~~~~~~~~~~~

编译时使用catkin_make -j1，会先生成States.h文件，再编译主程序。一次不行就多编译几次

MagicShawn, 2024-11-25:
> > LiDAR_IMU_Init/src/IMU_Processing.hpp：27：10：致命错误：lidar_imu_init/States.h：没有这样的文件或目录 #include <lidar_imu_init/States.h> ^~~~~~~~~~~~~~~~~~~~~~~~~
> 
> 编译时使用catkin_make -j1，会先生成States.h文件，再编译主程序。一次不行就多编译几次

非常感谢！

closed 2024-12-05 by MagicShawn