hku-mars/LiDAR_IMU_Init #116: 禾赛激光雷达pandar32XT标定报错

shixiaosongye, 2024-11-26:
[image]
你好，在运行标定时候，录制的数据包初始化完成后直接就报这个错误？是什么原因呢？我在使用官方给定的禾赛数据包标定时，初始化完成后，但最后迟迟没有出现标定结果，这个又是什么原因呢？

zfc-zfc, 2024-11-27:
> 错误

这个问题我们很久没遇到过了[手动捂脸] 方便给个bag吗 我这边尝试找找原因

85256638, 2024-12-05:
> ![微信图片_20241126130426](https://private-user-images.githubusercontent.com/110023919/389800165-915138ea-e642-4392-b3fe-c47ed4488d15.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzM0MTEwMjEsIm5iZiI6MTczMzQxMDcyMSwicGF0aCI6Ii8xMTAwMjM5MTkvMzg5ODAwMTY1LTkxNTEzOGVhLWU2NDItNDM5Mi1iM2ZlLWM0N2VkNDQ4OGQxNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIwNVQxNDU4NDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0yZmVmZjM5NTk0YWJkMWViMjY1MTFmMjgzNTg4MjMxNjc3MWY4ZDYyNTY5NDZlMDdkZjY0MmZlZGI5OTg4ZTlkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.cUYsp9LHV-OZMJX6i2eren8quVHa2htLLxSZHKeLtwg) 你好，在运行标定时候，录制的数据包初始化完成后直接就报这个错误？是什么原因呢？我在使用官方给定的禾赛数据包标定时，初始化完成后，但最后迟迟没有出现标定结果，这个又是什么原因呢？

请问佬用的什么IMU?我是XT-16,你怎么做IMU和激光雷达的硬同步了？

85256638, 2024-12-13:
@shixiaosongye 我也遇到了同样的错误，但是目前貌似解决了，解决方法如下，你可以试一试：

1. 使用Hesai的general SDK和ROS驱动，不要使用2代：
https://github.com/HesaiTechnology/HesaiLidar_General_ROS
https://github.com/HesaiTechnology/HesaiLidar_General_SDK

2. 查看你IMU的坐标轴和激光雷达的坐标轴，固定安装他们的时候尽量让他们尽量在同一个方向：
[image]
[image]

[image]

3. 注意你的IMU输出的角速度是多少！这个代码默认只支持IMU输出的RAD/S，如果你IMU输出的degree/S的话，你得修改代码重新编译

我按上面更改完之后，就没飘了。。。但是由于激光雷达和IMU不是硬同步，所以建图模糊。

shixiaosongye, 2024-12-14:
非常感谢，俺试一试谢谢~

 

  
 
 https://github.com/HesaiTechnology/HesaiLidar_General_ROS
 https://github.com/HesaiTechnology/HesaiLidar_General_SDK
 
 
 
 image.png (view on web)
 image.png (view on web)
 
  
image.png (view on web)

shixiaosongye, 2024-12-14:
我调整了IMU的频率到100hz，发现也是可以运行这个代码的，初始化很快，但标定的精度感觉不是很好。

85256638, 2024-12-15:
> 我调整了IMU的频率到100hz，发现也是可以运行这个代码的，初始化很快，但标定的精度感觉不是很好。

初始化很快不是很正常。。。二阶段标定飘了吗？

85256638, 2024-12-15:
> 我调整了IMU的频率到100hz，发现也是可以运行这个代码的，初始化很快，但标定的精度感觉不是很好。

另外请问你采用了我上面哪个建议能运行标定了？可以回复一下，这样也能帮助后来者。

shixiaosongye, 2024-12-18:
@85256638 非常感谢您的回复，我尝试了下对齐坐标系后，标定的依旧很漂，其中z方向加速度也不太对（理论上应该是9.7但我在标定时标出了7.多。。）目前还在找到原因 ；另外也想问下大佬这个跑slam时，因为禾赛数据格式要转换为velodyne，这个标定的结果在slam配置文件中需要变换吗？直接用跑出来的4X4矩阵内容即可呀？@zfc-zfc 也想问大佬如果坐标系不在同一方向这种运行标定可以吗？

shixiaosongye, 2024-12-18:
@85256638  我是IMU频率到100hz后就可以运行了，但禾赛的时间需要调整下，我也使用的系统时间，标定的经验是旋转慢一些，并且尽可能的空旷。标出来效果会好一些。但有一些问题是如果运行liosam这个标定结果怎么和liosam参数匹配起来呢@zfc-zfc？

> > 我调整了IMU的频率到100hz，发现也是可以运行这个代码的，初始化很快，但标定的精度感觉不是很好。
> 
> 另外请问你采用了我上面哪个建议能运行标定了？可以回复一下，这样也能帮助后来者。

1137472490, 2024-12-19:
> 我是IMU频率到100hz后就可以运行了，但禾赛的时间需要调整下，我也使用的系统时间，标定的经验是旋转慢一些，并且尽可能的空旷。标出来效果会好一些。但有一些问题是，如果运行liosam这个标定结果怎么和liosam参数匹配起来呢？
> 
> > > 我调整了IMU的频率到100hz，发现也是可以运行这个代码的，初始化很快，但标定的精度感觉不是很好。
> > 
> > 
> > 另外请问你采用了我上面哪个建议能运行标定了？可以回复一下，这样也能帮助后来者。

我用的也是PandarXT-32但是还是像你这样，什么包都不行，能否给个邮箱啥的，请教一下，拜托

85256638, 2024-12-21:
[image]

我的安装图如下：

[image]

不好意思，没有3D打印机，所以随便找了个硬纸盒子，然后背面拿螺丝钉固定了起来。。。看起来很廉价，等下周3D打印的底座到了之后在固定安装标定一下。。。

1. IMU和激光雷达的坐标系在ROS中和实际中是一致的。

2. 我的标定结果当时存到另外一个电脑上了，但是基本上标定出来得到的旋转矩阵R和平移移矩阵T和现实中激光雷达的坐标与IMU的坐标系相差不大。然后激光雷达和IMU的延迟大概是0.02s。。如果我没记错的话。 但是我建图有点模糊，估计是激光雷达没有和IMU硬同步。。

3. 关于把标定的数据绘制出图，我之前也回复过你：
[image]

如果还有问题，你私信我你的微信，我加你微信交流也可以。

1137472490, 2024-12-21:
> ![image](https://private-user-images.githubusercontent.com/3270742/397914853-f8255c23-4be8-43e5-8b1b-74ba1bdd29fd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ3ODc3MjksIm5iZiI6MTczNDc4NzQyOSwicGF0aCI6Ii8zMjcwNzQyLzM5NzkxNDg1My1mODI1NWMyMy00YmU4LTQzZTUtOGIxYi03NGJhMWJkZDI5ZmQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjFUMTMyMzQ5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTA3YTRmZDUyMzU2ODBhOTJlODEyMGM3YjJjMzY5ODY1YTczNGJiYTFiMTBjMDA2MDk0OWZjNGFiZmVhNTdhNCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.DKT7MIFZ2K_3LlNEEq8qf5X8-JlW9_r0ND-MKyJ0sTY)
> 
> @shixiaosongye 我的安装图如下：
> 
> ![image](https://private-user-images.githubusercontent.com/3270742/397914888-61f45d75-b626-4309-9e02-969143ba4fc5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ3ODc3MjksIm5iZiI6MTczNDc4NzQyOSwicGF0aCI6Ii8zMjcwNzQyLzM5NzkxNDg4OC02MWY0NWQ3NS1iNjI2LTQzMDktOWUwMi05NjkxNDNiYTRmYzUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjFUMTMyMzQ5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MWI5OTAyNjhiODk2YmJiNWZjMzJhMzA2ODllYjdkMzk1MGVmOWY1M2FhNTA5ZmYyZDQzMTBkNTUxZjMzODkzZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.oA9w0dpj9Zx5LbUEkNLJO7fSL-P7qs9wkkc-MSHTl9Y)
> 
> 不好意思，没有3D打印机，所以随便找了个硬纸盒子，然后背面拿螺丝钉固定了起来。。。看起来很廉价，等下周3D打印的底座到了之后在固定安装标定一下。。。
> 
> 1. IMU和激光雷达的坐标系在ROS中和实际中是一致的。
> 2. 我的标定结果当时存到另外一个电脑上了，但是基本上标定出来得到的旋转矩阵R和平移移矩阵T和现实中激光雷达的坐标与IMU的坐标系相差不大。然后激光雷达和IMU的延迟大概是0.02s。。如果我没记错的话。 但是我建图有点模糊，估计是激光雷达没有和IMU硬同步。。
> 3. 关于把标定的数据绘制出图，我之前也回复过你：
>    ![image](https://private-user-images.githubusercontent.com/3270742/397915113-b077ac52-91bc-4a55-97fc-77042fed7b6a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ3ODc3MjksIm5iZiI6MTczNDc4NzQyOSwicGF0aCI6Ii8zMjcwNzQyLzM5NzkxNTExMy1iMDc3YWM1Mi05MWJjLTRhNTUtOTdmYy03NzA0MmZlZDdiNmEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIyMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMjFUMTMyMzQ5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODY0YWVmNTdhYzEzMWJhMWFlZTFkNzc1NThmNWVlOWExOTBmNTdmN2ViZmMwMjhlMzZmMzAwZTE1NjBiY2Q2ZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.SQJ891bPYeV8AhEqMMDGaYT3_whKBsZDd-FxM5CId0U)
> 
> 如果还有问题，你私信我你的微信，我加你微信交流也可以。

[image]
您好我的微信是Kxy_599，我现在还是这样，很玄学，我在我的小车主控机上跑不了，在另一台电脑上又可以， 我分不出原因，希望可以细聊一下，万分感谢

HYF1011, 2025-01-03:
> > 我调整了IMU的频率到100hz，发现也是可以运行这个代码的，初始化很快，但标定的精度感觉不是很好。
> 
> 另外请问你采用了我上面哪个建议能运行标定了？可以回复一下，这样也能帮助后来者。

你好，请问你的16线雷达改了哪些代码啊，我的也是16线播放自己的数据包错误，录制数据包需要注意什么东西吗，如果可以可以分享一下你的数据包吗