hku-mars/LiDAR_IMU_Init #117: 同一个雷达，内置IMU，每次标定结果有差异

seven12767, 2024-12-16:
请问大佬，同一个场景，采集多组数据，每次标定结果都有差异，理论平移外参小于5cm，标定结果会大于10cm，请问标定时对采集数据的手法有要求吗？比如运动幅度、速度啥的？另外，我这边跑出来的线性加速度曲线很奇怪，lidar和imu差别很大，请问这个是运动激励不够嘛？
[image]

zfc-zfc, 2024-12-16:
> 请问大佬，同一个场景，采集多组数据，每次标定结果都有差异，理论平移外参小于5cm，标定结果会大于10cm，请问标定时对采集数据的手法有要求吗？比如运动幅度、速度啥的？另外，我这边跑出来的线性加速度曲线很奇怪，lidar和imu差别很大，请问这个是运动激励不够嘛？ ![acc](https://private-user-images.githubusercontent.com/71322283/395975694-13fafde2-20aa-4753-8870-7312795307d5.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQzMjg4NTQsIm5iZiI6MTczNDMyODU1NCwicGF0aCI6Ii83MTMyMjI4My8zOTU5NzU2OTQtMTNmYWZkZTItMjBhYS00NzUzLTg4NzAtNzMxMjc5NTMwN2Q1LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE2VDA1NTU1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVhYTMxNGNjYmUyNTZkYWI0ZDE3YTYxNDQ1YTIxNjQ5MzRhYzIzYThmNmM5NWMxODllNGE5OTQ5Mzk5YjZkNzYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.gVvFVUpgYE-uQQ2mZP-5VfdmoWzXrMebhGXAe-ZVJ7A)

对手法是有一定要求的，可以参考视频里面的方法，沿着三个轴画弧线。这里看起来，lidar的加速度noise有点大，可以尝试把acc_cov再调小一点；同时加速度的激励不太充分，可以加减速稍微明显一些。

seven12767, 2024-12-17:
感谢指导，还想请问大佬，acc_cov主要影响什么？我设置的acc_cov=2, gyr_cov=2，这两个值要设置多少比较合适？或者说根据哪些数据来调整这个值？我尝试acc_cov=0.5，lidar和imu的加速度曲线还是差别较大。另外，我用的是固态雷达，FOV为120°，请问这种情况下cut_frame_num设置多少比较合适？ @zfc-zfc