hku-mars/LiDAR_IMU_Init #118: 禾赛激光雷达16线雷达与阿路比LPMS-IG1 RS485IMU标定问题

HYF1011, 2024-12-25:
[image]
请问16线的与32线相比要改什么参数啊，imu不一样要改哪些地方。上面只改了imu的话题运行出来得出的问题。

zfc-zfc, 2025-01-02:
> ![微信图片_20241225113620](https://private-user-images.githubusercontent.com/178983559/398519285-6142778c-40f8-4753-a07d-0fad339f866c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU4MDAzNDEsIm5iZiI6MTczNTgwMDA0MSwicGF0aCI6Ii8xNzg5ODM1NTkvMzk4NTE5Mjg1LTYxNDI3NzhjLTQwZjgtNDc1My1hMDdkLTBmYWQzMzlmODY2Yy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEwMlQwNjQwNDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMDE0ZTk2NDQzZTJiNjZjZDJmNWE1NTYxMjA2NmYzMTlmNGY2MTQ4Yjk3OTg2M2FlODFjMzc2NDExMjYwOGQwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.xVXlhjzKeV-ZVCJXsFAmYDoXhunE_IaSaWwAK0z631c) 请问16线的与32线相比要改什么参数啊，imu不一样要改哪些地方。上面只改了imu的话题运行出来得出的问题。

理论上只需要把线数 scan_line 改掉就好。可以把参数表和rosbag上传网盘分享一下吗？

Ratarang, 2025-01-02:
[image]
可能是滤波器的extend_num过大，导致越界

Ratarang, 2025-01-02:
可能是LI_init.cpp里面Butter_filt函数的extend_num过大，导致后面数字越界了，我改成`butter.extend_num = std::min(static_cast<int>(signal_in.size() / 2), 10 * (butter.Coeff_size - 1));`就好了

HYF1011, 2025-01-03:
链接：https://pan.baidu.com/s/1IoXy3EChTsf0tEp4fEE8yQ 
提取码：a123

你好，这是我的数据包和相关雷达和imu产品说明书

HYF1011, 2025-01-03:
改了还是错

HYF1011, 2025-01-03:
改过后的，第一张图片里的Acceleration norm:0

 
image.png (view on web)

June-cy, 2025-05-06:
检查下ceres是不是2.0版本。我之前电脑装的ceres2.1版本，执行在执行优化时会崩溃，我换成2.0版本的ceres就好了