hku-mars/LiDAR_IMU_Init #127: 是否能通过录制的点云数据包来进行Lidar-Imu的标定工作？

Leslie-Jane, 2025-05-09:
当我用ouster 128线激光雷达(内置imu)录制了一个12s的点云数据包。然后启动roslaunch lidar_imu_init ouster.launch命令，再通过rosbag play 123.bag -r 0.5命令重新播放点云包时。终端中出现
[ WARN] [1746783618.087506295]: LI-Init not ready, no points stored.(问题不大，应该是没有启动点云文件)
Acceleration norm  : 10.0114
等待点云包全部播放完成之后。按理来说程序应该执行完成，但是好像卡在某一个环节了。

我只能通过ctrl+c终止程序，显示如下错误结果。

注：雷达录制点云包时是静止不动的(不知道对程序这个有没有影响)

Leslie-Jane, 2025-05-09:
> 当我用ouster 128线激光雷达(内置imu)录制了一个12s的点云数据包。然后启动roslaunch lidar_imu_init ouster.launch命令，再通过rosbag play 123.bag -r 0.5命令重新播放点云包时。终端中出现 [ WARN] [1746783618.087506295]: LI-Init not ready, no points [...]

上述卡在某一环节是如下图所示

[image]
错误结果如下图所示

[image]

Leslie-Jane, 2025-05-09:
> 当我用ouster 128线激光雷达(内置imu)录制了一个12s的点云数据包。然后启动roslaunch lidar_imu_init ouster.launch命令，再通过rosbag play 123.bag -r 0.5命令重新播放点云包时。终端中出现 [ WARN] [1746783618.087506295]: LI-Init not ready, no points [...]

必须得实时读取雷达数据，且雷达需要平移或者旋转才能标定成功吗？

dragon-bean001, 2025-12-03:
请问你最后解决这个问题了吗